raspibolt/raspibolt #1019: Raspibolt doesn't work anymore after system upgrade

Gilbog, 2022-05-12:
### Describe the issue
I installed Raspibolt 3 and everything worked fine. 

Today I did a `sudo apt update` and `sudo apt upgrade` according to the guide. 
This upgraded 
`Get:1 http://archive.raspberrypi.org/debian bullseye/main arm64 dphys-swapfile all 20100506-7+rpt1 [22.0 kB]`
`Get:2 http://archive.raspberrypi.org/debian bullseye/main arm64 raspi-config all 20220506 [30.4 kB]`
`Get:3 http://archive.raspberrypi.org/debian bullseye/main arm64 rpi-eeprom arm64 13.14-1 [1133 kB]`
The upgrade process exited normally.

But now the Raspibolt doesn't function anymore. I tried a reboot, but that didn't fix the problem. 

The System Overview doesn't appear any more, instead a flurry of error messages. Neither the lncli commands work anymore, they also result in errors.

**Errors after running `raspibolt` to start System Overview:**
`admin ~ ₿ raspibolt`

RaspiBolt 3: Sovereign Bitcoin full node
━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━
[lncli] rpc error: code = Unavailable desc = connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused"
[lncli] rpc error: code = Unavailable desc = connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused"
[lncli] rpc error: code = Unavailable desc = connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused"
[lncli] rpc error: code = Unavailable desc = connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused"
[lncli] rpc error: code = Unavailable desc = connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused"
[lncli] rpc error: code = Unavailable desc = connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused"
[lncli] rpc error: code = Unavailable desc = connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused"
[lncli] rpc error: code = Unavailable desc = connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused"
[lncli] rpc error: code = Unavailable desc = connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused"
[lncli] rpc error: code = Unavailable desc = connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused"
cpu temp: 62°C  tx: 10.0MiB    storage:   356G free     load: 1.36, 1.04, 0.93
up: 19 min      rx: 21.5MiB    2nd drive: none          available mem: 2484M
━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━
     .~~.   .~~.      ฿itcoin (mainnet)   up     Lightning (LND)     up
    '. \ ' ' / .'     v23.0.0                     Update!
     .~ .~~~..~.      Sync    OK [#736106]
    : .~.'／/~. :     Mempool 825 tx             ฿                  sat
   ~ (  ／ /_____~    Peers   12 (📥2 /📤10)     ⚡                 sat
  ( : ／____   ／ )                              ⏳               0 sat
   ~ .~ (  / ／. ~    Electrum            up     ∑                0 sat
    (  : '/／:  )     v0.9.7                     / channels
     '~ .~°~. ~'                                 Channel.db size: 205M
         '~'          Block Explorer      up
                      v3.3.0                     RTL                 up
                                                 v0.12.3

For others to connect to this lightning node
lncli connect

**Errors after running an `lncli` command:**
`admin ~ ₿ lncli walletbalance`
[lncli] rpc error: code = Unavailable desc = connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused"

### Location in guide
https://raspibolt.org/guide/raspberry-pi/system-configuration.html#system-update

### Environment
 - Raspberry Pi 4 with 4 GB
 - Raspberry Pi OS 64bit
 - Version: Debian GNU/Linux 11 (bullseye)
 - LND version 0.14.3-beta

Thanks for helping!

RooSoft, 2022-05-12:
Looks like LND is not running... have you tried a bit later? Bitcoin core and LND take some time to start.

Gilbog, 2022-05-12:
Thanks, RooSoft. But it's been over an hour since the reboot by now ... I doubt it could take that long? The node was fully synced, and the errors appeared _immediately_ after the upgrade, even before I did the reboot.

Should I perhaps have turned off LND before the system upgrade? But the guide doesn't say so. Could it be that the upgrade trashed LND somehow? Or overwrote some configurations?

RooSoft, 2022-05-13:
Normally an apt update doesn't require it to shutdown it as far as I know.

Maybe for, some reason, your LND folder probably got corrupt.

Can you see more information if you type that command?

```bash
sudo journalctl -t lnd
```

`If you don't have any channels open`, you might just try and rename the LND config folder and see how it regenerates.

Gilbog, 2022-05-14:
I looked at the last lines of `sudo journalctl -t lnd`, here they are:

May 14 09:03:36 raspibolt lnd[4970]: error starting REST proxy: listen tcp4 127.0.0.1:8080: bind: address al>
May 14 09:04:06 raspibolt lnd[4981]: Attempting automatic RPC configuration to bitcoind
May 14 09:04:06 raspibolt lnd[4981]: Automatically obtained bitcoind's RPC credentials
May 14 09:04:06 raspibolt lnd[4981]: 2022-05-14 09:04:06.248 [INF] LTND: Version: 0.14.3-beta commit=v0.14.3>
May 14 09:04:06 raspibolt lnd[4981]: 2022-05-14 09:04:06.248 [INF] LTND: Active chain: Bitcoin (network=main>
May 14 09:04:06 raspibolt lnd[4981]: 2022-05-14 09:04:06.250 [INF] RPCS: RPC server listening on 127.0.0.1:1>
May 14 09:04:06 raspibolt lnd[4981]: 2022-05-14 09:04:06.254 [ERR] LTND: gRPC proxy unable to listen on 127.>
May 14 09:04:06 raspibolt lnd[4981]: 2022-05-14 09:04:06.254 [ERR] LTND: Shutting down because error in main>
May 14 09:04:06 raspibolt lnd[4981]: 2022-05-14 09:04:06.263 [INF] LTND: Shutdown complete
May 14 09:04:06 raspibolt lnd[4981]: error starting REST proxy: listen tcp4 127.0.0.1:8080: bind: address al>
May 14 09:04:36 raspibolt lnd[4996]: Attempting automatic RPC configuration to bitcoind
May 14 09:04:36 raspibolt lnd[4996]: Automatically obtained bitcoind's RPC credentials
May 14 09:04:36 raspibolt lnd[4996]: 2022-05-14 09:04:36.609 [INF] LTND: Version: 0.14.3-beta commit=v0.14.3>
May 14 09:04:36 raspibolt lnd[4996]: 2022-05-14 09:04:36.609 [INF] LTND: Active chain: Bitcoin (network=main>
May 14 09:04:36 raspibolt lnd[4996]: 2022-05-14 09:04:36.611 [INF] RPCS: RPC server listening on 127.0.0.1:1>
May 14 09:04:36 raspibolt lnd[4996]: 2022-05-14 09:04:36.616 [ERR] LTND: gRPC proxy unable to listen on 127.>
May 14 09:04:36 raspibolt lnd[4996]: 2022-05-14 09:04:36.616 [ERR] LTND: Shutting down because error in main>
May 14 09:04:36 raspibolt lnd[4996]: 2022-05-14 09:04:36.625 [INF] LTND: Shutdown complete
May 14 09:04:36 raspibolt lnd[4996]: error starting REST proxy: listen tcp4 127.0.0.1:8080: bind: address al>
May 14 09:05:07 raspibolt lnd[5007]: Attempting automatic RPC configuration to bitcoind
May 14 09:05:07 raspibolt lnd[5007]: Automatically obtained bitcoind's RPC credentials
May 14 09:05:07 raspibolt lnd[5007]: 2022-05-14 09:05:07.026 [INF] LTND: Version: 0.14.3-beta commit=v0.14.3>
May 14 09:05:07 raspibolt lnd[5007]: 2022-05-14 09:05:07.026 [INF] LTND: Active chain: Bitcoin (network=main>
May 14 09:05:07 raspibolt lnd[5007]: 2022-05-14 09:05:07.028 [INF] RPCS: RPC server listening on 127.0.0.1:1>
May 14 09:05:07 raspibolt lnd[5007]: 2022-05-14 09:05:07.031 [ERR] LTND: gRPC proxy unable to listen on 127.>
May 14 09:05:07 raspibolt lnd[5007]: 2022-05-14 09:05:07.031 [ERR] LTND: Shutting down because error in main>
May 14 09:05:07 raspibolt lnd[5007]: 2022-05-14 09:05:07.041 [INF] LTND: Shutdown complete
May 14 09:05:07 raspibolt lnd[5007]: error starting REST proxy: listen tcp4 127.0.0.1:8080: bind: address al>

It looks like some process is repeating over and over but keeps failing. But I don't understand enough to interpret this technically. 

I do have a channel open, just a single experimental one, but it's quite big and I wouldn't want to lose the funds, obviously. 

I had set up the local backups from the Raspibolt 3 installation (https://raspibolt.org/guide/lightning/channel-backup.html#option-1-local-backup), and there there seem to be backups in the USB stick:

admin ~ ₿ ls -la /mnt/static-channel-backup-external
total 84
drwxr-xr-x 3 lnd  lnd  16384 Jan  1  1970  .
drwxr-xr-x 4 root root  4096 May  4 20:37  ..
-rwxr-xr-x 1 lnd  lnd    454 May  4 15:51  channel-20220504-155115.backup
-rwxr-xr-x 1 lnd  lnd    454 May  4 17:34  channel-20220504-173449.backup
-rwxr-xr-x 1 lnd  lnd    454 May  4 20:10  channel-20220504-201008.backup

Preferably I'd like to recover the channel as smoothly as possible. So not sure if I should proceed with your suggestion of renaming the directory?

VajraOfIndra, 2022-05-14:
Is `bitcoind` up and rtunning and fully sync? If so, try to restart LND `sudo systemctl restart lnd`. 

It happened to me yesterday when rebooting my node. A first known issue is that the blockchain takes time to sync if it was off for a few hours. A second one is that LND might take a long time to start if it compacts the database. But there is still an unresolved issue like I had yesterday that was solved by stopping and restarting LND.. but I don't know the cause of this. It'd be great to find it out one day.

Gilbog, 2022-05-14:
Yes, `bitcoind` is running and is fully synced.

A have tried to manually restart LND, but it won't. Seems like something is definitely broken in my system. It's not just a matter of waiting:
`sudo systemctl start lnd.service`
`sudo systemctl status lnd.service`
\* lnd.service - LND Lightning Network Daemon
     Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Sat 2022-05-14 15:19:16 BST; 3s ago
    Process: 2732 ExecStart=/usr/local/bin/lnd (code=exited, status=1/FAILURE)
   Main PID: 2732 (code=exited, status=1/FAILURE)
        CPU: 243ms

May 14 15:19:46 raspibolt systemd[1]: lnd.service: Scheduled restart job, restart counter is at 66.
May 14 15:19:46 raspibolt systemd[1]: Stopped LND Lightning Network Daemon.
May 14 15:19:46 raspibolt systemd[1]: Started LND Lightning Network Daemon.
May 14 15:19:50 raspibolt lnd[2746]: Attempting automatic RPC configuration to bitcoind
May 14 15:19:50 raspibolt lnd[2746]: Automatically obtained bitcoind's RPC credentials
May 14 15:19:50 raspibolt lnd[2746]: 2022-05-14 15:19:50.216 [INF] LTND: Version: 0.14.3-beta commit=v0.>
May 14 15:19:50 raspibolt lnd[2746]: 2022-05-14 15:19:50.216 [INF] LTND: Active chain: Bitcoin (network=>
May 14 15:19:50 raspibolt lnd[2746]: 2022-05-14 15:19:50.217 [INF] RPCS: RPC server listening on 127.0.0>
May 14 15:19:50 raspibolt lnd[2746]: 2022-05-14 15:19:50.220 [ERR] LTND: gRPC proxy unable to listen on >
May 14 15:19:50 raspibolt lnd[2746]: 2022-05-14 15:19:50.220 [ERR] LTND: Shutting down because error in >
May 14 15:19:50 raspibolt lnd[2746]: 2022-05-14 15:19:50.225 [INF] LTND: Shutdown complete
May 14 15:19:50 raspibolt lnd[2746]: error starting REST proxy: listen tcp4 127.0.0.1:8080: bind: addres>
May 14 15:19:50 raspibolt systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
May 14 15:19:50 raspibolt systemd[1]: lnd.service: Failed with result 'exit-code'.
lines 1-21/21 (END)

--

Should I reinstall the whole Raspibolt and use the seed phrase to re-create the old LND on-chain wallet and the backup to restore my one channel?

I'm hesitant, as I have no experience with restoring channels, and there are no instructions on restoring channels with the backup in the Raspibolt 3 guide.

Here is what I have in terms of channel backups (I followed the Raspibolt 3 guide to set this up to backup automatically on a USB stick in the Raspberry Pi).

admin ~ ₿ ls -la /mnt/static-channel-backup-external
total 84
drwxr-xr-x 3 lnd lnd 16384 Jan 1 1970 .
drwxr-xr-x 4 root root 4096 May 4 20:37 ..
-rwxr-xr-x 1 lnd lnd 454 May 4 15:51 channel-20220504-155115.backup
-rwxr-xr-x 1 lnd lnd 454 May 4 17:34 channel-20220504-173449.backup
-rwxr-xr-x 1 lnd lnd 454 May 4 20:10 channel-20220504-201008.backup

RooSoft, 2022-05-14:
Looks like RPC is listening to 127.0.0, which is not even a valid address.

Can you check your lnd.conf to see if you find it? If so, might be a good idea to change it to 127.0.0.1

Gilbog, 2022-05-14:
That's actually just a cut-off line. If I draw the terminal window a bit wider, it does show the full 127.0.0.1 etc., see: 

`sudo systemctl status lnd.service`
\* lnd.service - LND Lightning Network Daemon
     Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Sat 2022-05-14 20:38:47 BST; 26s ago
    Process: 7862 ExecStart=/usr/local/bin/lnd (code=exited, status=1/FAILURE)
   Main PID: 7862 (code=exited, status=1/FAILURE)
        CPU: 210ms

May 14 20:39:18 raspibolt lnd[7882]: Automatically obtained bitcoind's RPC credentials
May 14 20:39:18 raspibolt lnd[7882]: 2022-05-14 20:39:18.531 [INF] LTND: Version: 0.14.3-beta commit=v0.14.3-beta, build=prod>
May 14 20:39:18 raspibolt lnd[7882]: 2022-05-14 20:39:18.531 [INF] LTND: Active chain: Bitcoin (network=mainnet)
May 14 20:39:18 raspibolt lnd[7882]: 2022-05-14 20:39:18.538 [INF] RPCS: RPC server listening on 127.0.0.1:10009
May 14 20:39:18 raspibolt lnd[7882]: 2022-05-14 20:39:18.567 [ERR] LTND: gRPC proxy unable to listen on 127.0.0.1:8080
May 14 20:39:18 raspibolt lnd[7882]: 2022-05-14 20:39:18.567 [ERR] LTND: Shutting down because error in main method: error st>
May 14 20:39:18 raspibolt lnd[7882]: 2022-05-14 20:39:18.569 [INF] LTND: Shutdown complete
May 14 20:39:18 raspibolt lnd[7882]: error starting REST proxy: listen tcp4 127.0.0.1:8080: bind: address already in use
May 14 20:39:18 raspibolt systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
May 14 20:39:18 raspibolt systemd[1]: lnd.service: Failed with result 'exit-code'.

Gilbog, 2022-05-14:
But it seems that there indeed is **some problem related to 127.0.0.1:10009 being unavailable.** I just don't know enough to understand what the cause of the problem is and how to fix it. 

This repeated error is from my initial post, all the way up:

[lncli] rpc error: code = Unavailable desc = connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused"

VajraOfIndra, 2022-05-14:
What bonus programs did you install?

Also, could you run the following command and post the output:
`sudo netstat -tnlp | grep :8080`

Gilbog, 2022-05-14:
The output of `admin ~ ₿ sudo netstat -tnlp | grep :8080` is:
tcp        0      0 0.0.0.0:8080            0.0.0.0:*               LISTEN      945/node

From the Bonus section, I installed 

- System Overview
- Pimp the command line 
- Homer 
- Mempool
- Joinmarket

VajraOfIndra, 2022-05-14:
Ok, I'm not exactly sure what's going on but here is the issue I think.

This command should output something like that, indicating LND is using port 8080:
`tcp        0      0 127.0.0.1:8080          0.0.0.0:*               LISTEN      5313/lnd `

I think Homer or Mempool might interfere with LND, possibly due to a nginx config. A bug in our gudies somewhere.

Could you check is nginx, homer and mempool are running fine?

```
sudo systemctl status nginx
sudo systemctl status mempool
sudo systemctl status homer
```

@RooSoft wdyt the issue could be? I think I encountered that several times as well but I never took the time to really understand where the issue came from! (i usually just fixed the issue with a few random program restarts lol)

Gilbog, 2022-05-14:
Just to remind: _Everything_, including the Bonus programs, worked _perfectly_ before that system upgrade (see my intial post).

Here are the three status outputs. Even though Mempool and Homer say "running", I _cannot_ connect to them via a browser ("Unable to connect" error).

**nginx**
`admin ~ ₿ sudo systemctl status nginx`
[sudo] password for admin:
\* nginx.service - A high performance web server and a reverse proxy server
     Loaded: loaded (/lib/systemd/system/nginx.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Sat 2022-05-14 09:15:55 BST; 12h ago
       Docs: man:nginx(8)
    Process: 599 ExecStartPre=/usr/sbin/nginx -t -q -g daemon on; master_process on; (code=exited, status=1/FAILURE)
        CPU: 120ms

May 14 09:15:48 raspibolt systemd[1]: Starting A high performance web server and a reverse proxy server...
May 14 09:15:55 raspibolt nginx[599]: nginx: [emerg] host not found in upstream "mempool.space" in /etc/nginx/snippets/nginx->
May 14 09:15:55 raspibolt nginx[599]: nginx: configuration file /etc/nginx/nginx.conf test failed
May 14 09:15:55 raspibolt systemd[1]: nginx.service: Control process exited, code=exited, status=1/FAILURE
May 14 09:15:55 raspibolt systemd[1]: nginx.service: Failed with result 'exit-code'.
May 14 09:15:55 raspibolt systemd[1]: Failed to start A high performance web server and a reverse proxy server.
lines 1-13/13 (END)

**Mempool**
`admin ~ ₿ sudo systemctl status mempool`
\* mempool.service - mempool
     Loaded: loaded (/etc/systemd/system/mempool.service; enabled; vendor preset: enabled)
     Active: active (running) since Sat 2022-05-14 09:15:49 BST; 12h ago
   Main PID: 696 (node)
      Tasks: 15 (limit: 4163)
        CPU: 40min 3.160s
     CGroup: /system.slice/mempool.service
             `-696 /usr/bin/node --max-old-space-size=2048 dist/index.js

May 14 21:36:49 raspibolt node[696]: May 14 21:36:49 [696] DEBUG: Fetched transaction 4 / 10
May 14 21:36:49 raspibolt node[696]: May 14 21:36:49 [696] DEBUG: Fetched transaction 5 / 10
May 14 21:36:49 raspibolt node[696]: May 14 21:36:49 [696] DEBUG: Fetched transaction 6 / 10
May 14 21:36:49 raspibolt node[696]: May 14 21:36:49 [696] DEBUG: Fetched transaction 7 / 10
May 14 21:36:49 raspibolt node[696]: May 14 21:36:49 [696] DEBUG: Fetched transaction 8 / 10
May 14 21:36:49 raspibolt node[696]: May 14 21:36:49 [696] DEBUG: Fetched transaction 9 / 10
May 14 21:36:49 raspibolt node[696]: May 14 21:36:49 [696] DEBUG: Fetched transaction 10 / 10
May 14 21:36:50 raspibolt node[696]: May 14 21:36:50 [696] DEBUG: Mempool blocks calculated in 0.085 seconds
May 14 21:36:50 raspibolt node[696]: May 14 21:36:50 [696] DEBUG: New mempool size: 7422 Change: 10
May 14 21:36:50 raspibolt node[696]: May 14 21:36:50 [696] DEBUG: Mempool updated in 0.236 seconds

**Homer**
`admin ~ ₿ sudo systemctl status homer`
\* homer.service - Homer
     Loaded: loaded (/etc/systemd/system/homer.service; enabled; vendor preset: enabled)
     Active: active (running) since Sat 2022-05-14 09:15:48 BST; 12h ago
   Main PID: 597 (npm run serve)
      Tasks: 23 (limit: 4163)
        CPU: 49.130s
     CGroup: /system.slice/homer.service
             |-597 npm run serve
             |-937 sh -c vue-cli-service serve
             `-945 node /home/homer/homer/node_modules/.bin/vue-cli-service serve

May 14 14:46:47 raspibolt npm[945]: <s> [webpack.Progress] 95% emitting HtmlWebpackPlugin
May 14 14:46:47 raspibolt npm[945]: <s> [webpack.Progress] 95% emitting vue-cli:pwa-html-plugin
May 14 14:46:47 raspibolt npm[945]: <s> [webpack.Progress] 95% emitting CopyPlugin
May 14 14:46:48 raspibolt npm[945]: <s> [webpack.Progress] 98% after emitting
May 14 14:46:48 raspibolt npm[945]: <s> [webpack.Progress] 98% after emitting CopyPlugin
May 14 14:46:48 raspibolt npm[945]:  DONE  Compiled successfully in 27701ms14:46:48
May 14 14:46:48 raspibolt npm[945]: <s> [webpack.Progress] 100%
May 14 14:46:48 raspibolt npm[945]:   App running at:
May 14 14:46:48 raspibolt npm[945]:   - Local:   http://localhost:8080
May 14 14:46:48 raspibolt npm[945]:   - Network: http://192.168.1.205:8080

VajraOfIndra, 2022-05-14:
It's nginx which has some issues with Mempool after a reboot. It's a known issue #919 that we didn't fix yet.

Restart nginx with :
`sudo systemctl restart nginx`

Wait a few minutes and check again the status of these programs.

Gilbog, 2022-05-14:
This fixed a part of the problem: I can now access Mempool and Homer in the browser.

But RTL (and LND, presumably) still does not work, and the System Overview also doesn't work yet:

[image]

Here are the new status outputs:

**nginx**
`admin ~ ₿ sudo systemctl status nginx`
\* nginx.service - A high performance web server and a reverse proxy server
     Loaded: loaded (/lib/systemd/system/nginx.service; enabled; vendor preset: enabled)
     Active: active (running) since Sat 2022-05-14 21:57:22 BST; 1min 39s ago
       Docs: man:nginx(8)
    Process: 10470 ExecStartPre=/usr/sbin/nginx -t -q -g daemon on; master_process on; (code=exited, status=0/SUCCESS)
    Process: 10471 ExecStart=/usr/sbin/nginx -g daemon on; master_process on; (code=exited, status=0/SUCCESS)
   Main PID: 10472 (nginx)
      Tasks: 5 (limit: 4163)
        CPU: 160ms
     CGroup: /system.slice/nginx.service
             |-10472 nginx: master process /usr/sbin/nginx -g daemon on; master_process on;
             |-10473 nginx: worker process
             |-10474 nginx: worker process
             |-10475 nginx: worker process
             `-10476 nginx: worker process

May 14 21:57:22 raspibolt systemd[1]: Starting A high performance web server and a reverse proxy server...
May 14 21:57:22 raspibolt systemd[1]: Started A high performance web server and a reverse proxy server.

**Mempool**
`admin ~ ₿ sudo systemctl status mempool`
\* mempool.service - mempool
     Loaded: loaded (/etc/systemd/system/mempool.service; enabled; vendor preset: enabled)
     Active: active (running) since Sat 2022-05-14 09:15:49 BST; 12h ago
   Main PID: 696 (node)
      Tasks: 15 (limit: 4163)
        CPU: 42min 47.410s
     CGroup: /system.slice/mempool.service
             `-696 /usr/bin/node --max-old-space-size=2048 dist/index.js

May 14 22:02:47 raspibolt node[696]: May 14 22:02:47 [696] DEBUG: Fetched transaction 13 / 19
May 14 22:02:47 raspibolt node[696]: May 14 22:02:47 [696] DEBUG: Fetched transaction 14 / 19
May 14 22:02:47 raspibolt node[696]: May 14 22:02:47 [696] DEBUG: Fetched transaction 15 / 19
May 14 22:02:47 raspibolt node[696]: May 14 22:02:47 [696] DEBUG: Fetched transaction 16 / 19
May 14 22:02:47 raspibolt node[696]: May 14 22:02:47 [696] DEBUG: Fetched transaction 17 / 19
May 14 22:02:48 raspibolt node[696]: May 14 22:02:48 [696] DEBUG: Fetched transaction 18 / 19
May 14 22:02:48 raspibolt node[696]: May 14 22:02:48 [696] DEBUG: Fetched transaction 19 / 19
May 14 22:02:48 raspibolt node[696]: May 14 22:02:48 [696] DEBUG: Mempool blocks calculated in 0.099 seconds
May 14 22:02:48 raspibolt node[696]: May 14 22:02:48 [696] DEBUG: New mempool size: 7748 Change: 19
May 14 22:02:48 raspibolt node[696]: May 14 22:02:48 [696] DEBUG: Mempool updated in 1.808 seconds
... 

**Homer**
`admin ~ ₿ sudo systemctl status homer`
\* homer.service - Homer
     Loaded: loaded (/etc/systemd/system/homer.service; enabled; vendor preset: enabled)
     Active: active (running) since Sat 2022-05-14 09:15:48 BST; 12h ago
   Main PID: 597 (npm run serve)
      Tasks: 23 (limit: 4163)
        CPU: 49.218s
     CGroup: /system.slice/homer.service
             |-597 npm run serve
             |-937 sh -c vue-cli-service serve
             `-945 node /home/homer/homer/node_modules/.bin/vue-cli-service serve

May 14 14:46:47 raspibolt npm[945]: \<s> [webpack.Progress] 95% emitting HtmlWebpackPlugin
May 14 14:46:47 raspibolt npm[945]: \<s> [webpack.Progress] 95% emitting vue-cli:pwa-html-plugin
May 14 14:46:47 raspibolt npm[945]: \<s> [webpack.Progress] 95% emitting CopyPlugin
May 14 14:46:48 raspibolt npm[945]: \<s> [webpack.Progress] 98% after emitting
May 14 14:46:48 raspibolt npm[945]: \<s> [webpack.Progress] 98% after emitting CopyPlugin
May 14 14:46:48 raspibolt npm[945]:  DONE  Compiled successfully in 27701ms14:46:48
May 14 14:46:48 raspibolt npm[945]: \<s> [webpack.Progress] 100%
May 14 14:46:48 raspibolt npm[945]:   App running at:
May 14 14:46:48 raspibolt npm[945]:   - Local:   http://localhost:8080
May 14 14:46:48 raspibolt npm[945]:   - Network: http://192.168.1.205:8080

VajraOfIndra, 2022-05-14:
Ok, now disable and stop Homer for now. And restart LND.

```
sudo systemctl disable homer
sudo systemctl disable homer
sudo systemctl restart lnd
```

I think LND would work then..

Gilbog, 2022-05-14:
You wrote twice `disable`. Which is the correct order, if it matters? First `stop`, then `disable `or vice versa?

VajraOfIndra, 2022-05-14:
> You wrote twice `disable`. Which is the correct order, if it matters? First `stop`, then `disable` or vice versa?

Interesting question. I always assumed it was 1) stop and 2) disable but I saw it done 1) disable 2) stop recently and I believe it might be better for some edge cases. tk;dr: I don't really know :)

Gilbog, 2022-05-14:
**YESSS!** You rock, Indra, thank you so much! The miracle happened, LND works again!

I used this order:
```
sudo systemctl disable homer
sudo systemctl stop homer
sudo systemctl restart lnd
```
I'm so glad I didn't have to do any channel restore fiddling around, sweating about losing my sats. 

For the future, ... 

- should I better **not touch Homer** any more (I don't need it, really)?
- should I better **not do any system upgrades** any more, to not risk such a breakdown again?

Again, thanks a lot!

VajraOfIndra, 2022-05-14:
I think you can start Homer again now (first enable then start) and it should use the right port 4091.

I have Homer running with LND and Mempool, no problems excpet if I need to restart the node.

System upgardes are very important, continue to do them.
You could uninstall Homer for sure but it;s such a cool homepage :heart: 

Just bear with us and will try to find the root cause of these issues and fix them asap. In the meantime if you need to reboot, maybe stop and disable homer and mempool first, reboot, check that bitcoin and lnd work and then start mempool and homer again.

Gilbog, 2022-05-14:
Thanks so much for your advice. While it feels awful when something breaks down, your money is stuck in there, and you're helpless because you understand how it works, it's such a relief when it works again -- almost happier now than before :)

I really appreciate all your time helping me!

closed 2022-05-14 by Gilbog

RooSoft, 2022-05-15:
> Interesting question. I always assumed it was 1) stop and 2) disable but I saw it done 1) disable 2) stop recently and I believe it might be better for some edge cases. tk;dr: I don't really know :)

Stop and Disable are completely unrelated... stop is for the current instance running, and disable will prevent it from starting after a reboot. The order in which you run those commands doesn't matter.

Gilbog, 2022-05-15:
Thanks for clarifying this, RooSoft!

There still is one more problem with my Raspibolt: Right now, 
- the mempool shown on https://mempool.space/ is ca. **11,000 transactions** / 205 MB, which is also what my other node shows (mempool on a Raspilbitz)
- but mempool on my Raspibolt only shows about **2,600 transactions** / 42 MB

This is **more than 12 hours** after the Raspibolt has been online again **and fully synced,** after its full functionality was restored with the above steps.

VajraOfIndra, 2022-05-15:
> There still is one more problem with my Raspibolt: Right now, ...

It's better to keep to a single issue/topic per thread and to open a new one for a new issue. It's easier for future users to find answers to their questions when looking at these archived threads :)

If you don't mind, could you open a new issue with these questions..? :+1:

Gilbog, 2022-05-15:
Yes, makes sense ... I just reposted as a new issue, thanks.