raspibolt/raspibolt #1020: Mempool app doesn't show full mempool

Gilbog, 2022-05-15:
(This is a follow up to bug report https://github.com/raspibolt/raspibolt/issues/1019)

There still is one more problem with my Raspibolt. Right now, ...

 - the mempool shown on https://mempool.space/ is ca. **11,000 transactions** / 205 MB, which is also what my other node shows (mempool on a Raspiblitz)
 - but mempool on my Raspibolt only shows about **2,600 transactions** / 42 MB

This is more than 18 hours after the Raspibolt has been online again and fully synced, after its full functionality was restored with the above steps, so I'm quite sure this is not just a sync issue that would resolve itself over time. 

I checked with tail -f /home/bitcoin/.bitcoin/debug.log and I notice that there are a lot of "socks5" connection failures. Here is the output:

2022-05-15T08:45:43Z UpdateTip: new best=00000000000000000002a0d851db6606b789196ef7fefcf1f65352cca5ba93f9 height=736457 version=0x20800004 log2_work=93.520025 tx=733729208 date='2022-05-15T08:45:23Z' progress=1.000000 cache=153.1MiB(1153691txo)
2022-05-15T08:52:33Z Socks5() connect to 2a06:4944:ffd:a700:4d3d:efa2:fa58:240a:8333 failed: general failure
2022-05-15T08:52:59Z UpdateTip: new best=000000000000000000073796afafd121158b183c251addbe369ddefbc068577f height=736458 version=0x20c00004 log2_work=93.520038 tx=733730407 date='2022-05-15T08:48:04Z' progress=0.999999 cache=154.0MiB(1160661txo)
2022-05-15T08:53:00Z BlockUntilSyncedToCurrentChain: txindex is catching up on block notifications
2022-05-15T08:57:45Z UpdateTip: new best=00000000000000000004646fb1c36258bcb6840b45ae8d0b828388172c235936 height=736459 version=0x20c00004 log2_work=93.520052 tx=733731631 date='2022-05-15T08:52:37Z' progress=0.999999 cache=154.8MiB(1167316txo)
2022-05-15T08:57:52Z Socks5() connect to 2001:ac8:23:3c:2d7::1:8333 failed: general failure
2022-05-15T09:00:59Z Socks5() connect to 170.75.165.248:39388 failed: general failure
2022-05-15T09:14:20Z UpdateTip: new best=00000000000000000001de292462033547f76d89e56daf7ab8e290e4125878f7 height=736460 version=0x20a00004 log2_work=93.520065 tx=733734385 date='2022-05-15T09:13:06Z' progress=1.000000 cache=155.6MiB(1174228txo)
2022-05-15T09:16:43Z Socks5() connect to 178.142.29.32:8333 failed: general failure
2022-05-15T09:17:07Z UpdateTip: new best=00000000000000000008cf6249ac4ee3b84034ffa6af997e0470541222f70ea2 height=736461 version=0x2001c000 log2_work=93.520079 tx=733735428 date='2022-05-15T09:16:20Z' progress=1.000000 cache=156.4MiB(1180116txo)

## Environment
- Raspberry Pi 4 with 4 GB
- Raspberry Pi OS 64bit
- Version: Debian GNU/Linux 11 (bullseye)

kristapsk, 2022-05-18:
Try running `bitcoin-cli getmempoolinfo`. There will be `"usage"` which shows current memory usage of mempool and `"maxmempool"`, which shows maximum allowed memory usage for mempool. Default is 300 MiB. Can be changed with `maxmempool` `bitcoin.conf` setting.

Gilbog, 2022-05-18:
Thanks, all, for the help. In the meantime, I had an electricity outage in the house and that messed up everything again. I managed to use the tricks from the prior bug #1019 to get everything going again. 

Now, with the mempool almost empty, both Raspibolt and Raspiblitz show about the same amount of transactions, so whatever it was, I guess we can consider it fixed for now ;)

I'll close this.

closed 2022-05-18 by Gilbog

VajraOfIndra, 2022-05-18:
> Thanks, all, for the help. In the meantime, I had an electricity outage in the house and that messed up everything again.

To avoid this, you could consider using a [UPS](https://plebnet.wiki/wiki/UPS). We even have a draft guide to connect the Pi to the UPS for automatic shutdown when the battery is low: https://github.com/raspibolt/raspibolt/blob/5d728d2f468787804983f21e27fc6e8d9b672fcc/bonus/raspberry-pi/ups-nut.md

> Now, with the mempool almost empty, both Raspibolt and Raspiblitz show about the same amount of transactions, so whatever it was, I guess we can consider it fixed for now ;)

Glad to see it's fixed for now! :)

Gilbog, 2022-05-18:
Thanks for the UPS tip. I did at some point look into this, but frankly, I just don't know how to go about setting it up. The UPS hardware from the links in the wiki are mostly unavailable. Only the UK model is available. If I bought something else (and there is other hardware available) I'd be afraid that I couldn't set it up. Also it's pretty expensive. Finally, I had noticed, that the Raspiblitz is surprisingly resistant to power failures: it just reboots and works, including the lightning channels. So I felt a bit less urgency to explore that UPS topic further.

kristapsk, 2022-05-18:
> If I bought something else (and there is other hardware available) I'd be afraid that I couldn't set it up.

Even any UPS not connected to your node via USB data link will be helpful in a lot of cases (short time power failures). That's what I have at home actually (APC Back-UPS ES 550, was not too expensive, around 100 EUR IIRC). No special setup needed, just plug it in and that's all.

Gilbog, 2022-05-18:
True, just for bridging a short term outage, that makes sense. Yesterday's outage lasted ca. 45 min. I've had others that just lasted a minute. Those would probably be "bridgeable". I'll explore this sometime (when Bitcoin makes me feel a bit richer & more inclined to spend more money on hardware ;)

kristapsk, 2022-05-18:
Also, stuff like your router and ARM boxes like RPi are not very power hungry, so can last for some time with decent UPS.