raspibolt/raspibolt #1027: stuck at Security Step 2

kwikslvr, 2022-05-24:
### Describe the issue
I've setup everything in the guide to the point where I can successfully login with SSH keys as an admin on my Pi.  Started trying to work through the Uncomplicated Firewall setup, but when I enter the first command (sudo apt install ufw), I'm asked to enter my admin password.  For some reason it won't accept my password, despite typing it in perfectly, so I've hit a wall.  I thought the point of setting up the SSH keys was both to increase security, and eliminate the need to have to keep using my admin password.

### Location in guide
https://raspibolt.org/guide/raspberry-pi/security.html#enabling-the-uncomplicated-firewall

[image]

### Environment
 - Hardware platform = Raspberry Pi 4
 - Operating system = Raspberry Pi OS Lite (64bit)
 - Version = Debian GNU/Linux 11 (bullseye)

DanielPX1, 2022-05-24:
I guess it is simple answer, but I think you mistyped your password during first set up. You are really in a early stage of set up, so perhaps wipe all and start from scratch?

I HIGHLY RECOMMEND generating admin password in password manager, saving it there and then copying it. You will never make a mistake like that.  It is really bulletproof for these types of things

kwikslvr, 2022-05-28:
Update: wiped and started from scratch, and successfully made it through the Security setup steps this time.

closed 2022-05-28 by kwikslvr