raspibolt/raspibolt #1032: Bitcoin Core install issues

kwikslvr, 2022-05-29:
### Describe the issue
I'm at the installation step of Bitcoin client, have downloaded the latest Bitcoin Core binary along with the hash and signatures.  Reference checksum was ok, and signature check said there were good signatures, but with warnings.  I went ahead and extracted and installed Bitcoin Core anyway, but it fails the version check.  Not sure if I should try to proceed with Steps 2a and following.

### Location in guide
https://raspibolt.org/guide/bitcoin/bitcoin-client.html

### Screenshots
Here's are some of the signature warnings:
[image]

Here's the error when I tried to check the version after installing:
[image]

I tried to change to the bitcoin-23.0/bin/ directory where I can clearly see the file "bitcoind" listed (as shown in the screenshot) but it came back with the same error "no such file or directory"

### Environment
 - Hardware platform: Raspberry Pi 4
 - Operating system: Raspberry Pi OS 64bit Lite
 - Version: Debian GNU/Linux 11 (bullseye)

kwikslvr, 2022-06-05:
Any suggestions for how to resolve the issues I reported?  Can I safely ignore the warnings on the signatures and assume the bitcoin core client I downloaded is safe to use?  And why doesn't the version check work?  If it's a simple noob error, please advise - I don't know linux well enough to troubleshoot it myself.

zynos, 2022-06-10:
I installed btc core 23.0 yesterday on my raspi and it worked.
Why did you use the arm-linux version and not the suggested version from the guide?
`$ tar -xvf bitcoin-23.0-aarch64-linux-gnu.tar.gz`
I dont know if that is a problem but maybe you can try that version instead

kwikslvr, 2022-06-11:
@zynos I was going by the instruction in the guide to download the latest version in case there had been an update, as it says here: 
[image]

But looking at it again, I see the latest version is in fact 23.0, so I'll retry it like you did.  Thanks for the suggestion!

kwikslvr, 2022-06-11:
@zynos I went ahead and took your suggestion and just downloaded the version listed in the guide, and when I checked the signatures, got similar warnings as last time.

[image]

But I went ahead and installed the bitcoin core client and this time it worked as expected, so will close this issue.  Still not sure if those warnings should be heeded though - makes this noob a bit uncertain.  If they're not important, I feel like the guide should mention that they can be safely ignored.

closed 2022-06-11 by kwikslvr