raspibolt/raspibolt #1035: [BUG] TimeoutSec=300 might not be adequate and can lead to database corruption

jturning, 2022-06-02:
If you're running bitcoind with less memory, and or slower hard disks TimeoutSec=300 might not be long enough. And when bitcoind gets killed I've run into utxo db corruption and a lot of down time with -reindex-chainstate. I have encountered this with a 4GB Pi 4 board and external hard disk (now switched to SSD). And I also encountered this with a Proxmox VM on a Dell Poweredge server running on a single drive with 6 processors and 6 GB or ram (changed to 12 procs and 8 GB of ram until synced). So I've changed mine to TimeoutSec=infinity. There might be a more elegant way of handling this with systemd parameters giving it extra shutdown time, or put a note in the docs to up the value or use infinity if using slower drives or limited hardware in order to avoid database corruption on shutdown. 

Also in bitcoin.service we have it set to Restart=on-failure, but this is overridden by the statements in lnd.service and electrs.service as bitcoind will continuously restart and not respect the parameter in bitcoin.service. If you're trying to manually make sure bitcoind shuts down to power off the server or reboot you might be surprised by it continuously restarting. And bitcoind doesn't appear to respond to a shutdown while in the startup phase.

VajraOfIndra, 2022-08-09:
Yes, our present-day systemd service files have that small bug that makes bitcoind restart when one is trying to shut it down... this is not ideal and we should review the service file options to optimize the dependencies between our main program.

Anyone would be up for the task..?! I believe it's about selecting the right options (`Wants=` vs R`equires=` vs `Before=` vs `After=`) for the expected use cases (but also thinking about any edge cases).. probably while trying to prioritize DB corruption prevention.

Resource: e.g. this [intro ](https://access.redhat.com/documentation/en-us/red_hat_enterprise_linux/8/html/configuring_basic_system_settings/assembly_working-with-systemd-unit-files_configuring-basic-system-settings)to systemd service file configuration, notably section "17.3. Important [Unit] section options"

---

Not sure about the TimeoutSec issue, has anyone any comments on that one?

VajraOfIndra, 2022-10-08:
> If you're running bitcoind with less memory, and or slower hard disks TimeoutSec=300 might not be long enough. 

What value would have been ok for you? would using 600 or 900 seconds be enough for slower drives, low resource VMs etc? 
Is there actually a downside to have a large value..?

Did you encounter similar issues with LND? (it has a value of 240 sec for now)

> If you're trying to manually make sure bitcoind shuts down to power off the server or reboot you might be surprised by it continuously restarting.

See this [comment ](https://github.com/raspibolt/raspibolt/issues/848#issuecomment-1256046700) in #848 .
I propose to use `PartOf` rather than `Wants`.

My personal preference would be for a conservative handling of Bitcoin Core start and close behaviour, _i.e._:
- Start the dependent programs (Electrs, Explorer, LND etc) when Bitcoin Core starts
- When starting Bitcoin Core, if the dependent programs can't start (e.g. not installed), Bitcoin Core should still starts.
- The dependent prograns should start only AFTER Bitcoin Core has started
- Stop the dependent programs if Bitcoin Core stops
- If a dependent program is started but Bitcoin Core is not, do NOT start Bitcoin Core

The above goals would be fulfilled with the follwing systemd service files:

`bitcoind.service`:

> [Unit]
> After=network.target

`electrs.service`:

> [Unit]
> After=bitcoind.service
> PartOf=bitcoind.service

`btcrpcexplorer.service`:

> [Unit]
> After=bitcoind.service
> PartsOf=bitcoind.service

`lnd.service`:

> [Unit]
> After=bitcoind.service
> PartsOf=bitcoind.service

`rtl.service`:

> [Unit]
> After=lnd.service
> PartsOf=lnd.service
 
It'd be interesting to review the `Restart`, `RestartSec` and `TimeoutSec` options as well.

closed 2022-12-14 by github-actions[bot]