raspibolt/raspibolt #1041: [BUG] Installing Specter Desktop

fwittingham, 2022-06-25:
### Describe the bug
I followed all steps in the bonus section Installing Specter Desktop but when installing in the console I stuck with this error:
ERROR: Package 'cryptoadvance-specter' requires a different Python: 3.9.2 not in '<3.9,>=3.6'

### Location in guide
Link to the corresponding page and section on <https://raspibolt.org> <https://raspibolt.org/guide/bonus/bitcoin/specter-desktop.html#specter-desktop>

### Expected behavior
The application communicates directly with a Bitcoin full node via its RPC interface.

### Screenshots
<>
### Environment
 - Hardware platform: Raspberry Pi 4, 4GB
 - Operating system: Raspberry Pi OS 64bit
 - Version: Description:    Debian GNU/Linux 11 (bullseye)

### Additional context

specter@RaspiBolt:~/specter-desktop $ sed -i "s/\(version=\"\).*\(\",\)/\1${VERSION}\2/g; " setup.py
specter@RaspiBolt:~/specter-desktop $ /home/specter/.env/bin/python3 -m pip install --upgrade .
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /home/specter/specter-desktop
  Preparing metadata (setup.py) ... done
Collecting base58==2.0.1
  Using cached https://www.piwheels.org/simple/base58/base58-2.0.1-py3-none-any.whl (4.3 kB)
Collecting bitbox02==5.3.0
  Using cached https://www.piwheels.org/simple/bitbox02/bitbox02-5.3.0-py3-none-any.whl (80 kB)
Collecting certifi==2019.9.11
  Using cached https://www.piwheels.org/simple/certifi/certifi-2019.9.11-py2.py3-none-any.whl (154 kB)
Collecting cffi==1.14.3
  Using cached cffi-1.14.3.tar.gz (470 kB)
  Preparing metadata (setup.py) ... done
Collecting chardet==3.0.4
  Using cached https://www.piwheels.org/simple/chardet/chardet-3.0.4-py2.py3-none-any.whl (133 kB)
Collecting click==7.1.2
  Using cached https://www.piwheels.org/simple/click/click-7.1.2-py2.py3-none-any.whl (84 kB)
Collecting cryptography==3.3.2
  Using cached cryptography-3.3.2-cp36-abi3-manylinux2014_aarch64.whl (2.6 MB)
Collecting daemonize==2.5.0
  Using cached https://www.piwheels.org/simple/daemonize/daemonize-2.5.0-py2.py3-none-any.whl (5.2 kB)
Collecting ecdsa==0.16.1
  Using cached https://www.piwheels.org/simple/ecdsa/ecdsa-0.16.1-py2.py3-none-any.whl (104 kB)
Collecting embit==0.1.3
  Using cached https://www.piwheels.org/simple/embit/embit-0.1.3-py3-none-any.whl (1.4 MB)
Collecting flask-cors==3.0.8
  Using cached https://www.piwheels.org/simple/flask-cors/Flask_Cors-3.0.8-py2.py3-none-any.whl (14 kB)
Collecting flask-login==0.5.0
  Using cached https://www.piwheels.org/simple/flask-login/Flask_Login-0.5.0-py2.py3-none-any.whl (16 kB)
Collecting flask==1.1.2
  Using cached https://www.piwheels.org/simple/flask/Flask-1.1.2-py2.py3-none-any.whl (94 kB)
Collecting flask_wtf==0.14.3
  Using cached https://www.piwheels.org/simple/flask-wtf/Flask_WTF-0.14.3-py2.py3-none-any.whl (13 kB)
Collecting hidapi==0.7.99.post21
  Using cached hidapi-0.7.99.post21.tar.gz (51 kB)
  Preparing metadata (setup.py) ... done
Collecting hwi==2.0.1
  Using cached https://www.piwheels.org/simple/hwi/hwi-2.0.1-py3-none-any.whl (233 kB)
Collecting idna==2.10
  Using cached https://www.piwheels.org/simple/idna/idna-2.10-py2.py3-none-any.whl (58 kB)
Collecting importlib_metadata==2.0.0
  Using cached https://www.piwheels.org/simple/importlib-metadata/importlib_metadata-2.0.0-py2.py3-none-any.whl (31 kB)
Collecting itsdangerous==1.1.0
  Using cached https://www.piwheels.org/simple/itsdangerous/itsdangerous-1.1.0-py2.py3-none-any.whl (16 kB)
Collecting jinja2==2.11.3
  Using cached https://www.piwheels.org/simple/jinja2/Jinja2-2.11.3-py2.py3-none-any.whl (125 kB)
Collecting libusb1==1.8
  Using cached https://www.piwheels.org/simple/libusb1/libusb1-1.8-py3-none-any.whl (49 kB)
Requirement already satisfied: markupsafe==1.1.1 in /home/specter/.env/lib/python3.9/site-packages/MarkupSafe-1.1.1-py3.9-linux-aarch64.egg (from cryptoadvance.specter==1.3.0) (1.1.1)
Requirement already satisfied: mnemonic==0.18 in /home/specter/.env/lib/python3.9/site-packages/mnemonic-0.18-py3.9.egg (from cryptoadvance.specter==1.3.0) (0.18)
Requirement already satisfied: noiseprotocol==0.3.1 in /home/specter/.env/lib/python3.9/site-packages/noiseprotocol-0.3.1-py3.9.egg (from cryptoadvance.specter==1.3.0) (0.3.1)
Requirement already satisfied: pbkdf2==1.3 in /home/specter/.env/lib/python3.9/site-packages/pbkdf2-1.3-py3.9.egg (from cryptoadvance.specter==1.3.0) (1.3)
Requirement already satisfied: pgpy==0.5.3 in /home/specter/.env/lib/python3.9/site-packages/PGPy-0.5.3-py3.9.egg (from cryptoadvance.specter==1.3.0) (0.5.3)
Requirement already satisfied: protobuf==3.13.0 in /home/specter/.env/lib/python3.9/site-packages/protobuf-3.13.0-py3.9.egg (from cryptoadvance.specter==1.3.0) (3.13.0)
Requirement already satisfied: psutil==5.7.3 in /home/specter/.env/lib/python3.9/site-packages/psutil-5.7.3-py3.9-linux-aarch64.egg (from cryptoadvance.specter==1.3.0) (5.7.3)
Requirement already satisfied: pyaes==1.6.1 in /home/specter/.env/lib/python3.9/site-packages/pyaes-1.6.1-py3.9.egg (from cryptoadvance.specter==1.3.0) (1.6.1)
Requirement already satisfied: pyasn1==0.4.8 in /home/specter/.env/lib/python3.9/site-packages/pyasn1-0.4.8-py3.9.egg (from cryptoadvance.specter==1.3.0) (0.4.8)
Requirement already satisfied: pycparser==2.20 in /home/specter/.env/lib/python3.9/site-packages/pycparser-2.20-py3.9.egg (from cryptoadvance.specter==1.3.0) (2.20)
Requirement already satisfied: pyopenssl==20.0.1 in /home/specter/.env/lib/python3.9/site-packages/pyOpenSSL-20.0.1-py3.9.egg (from cryptoadvance.specter==1.3.0) (20.0.1)
Requirement already satisfied: pyserial==3.4 in /home/specter/.env/lib/python3.9/site-packages/pyserial-3.4-py3.9.egg (from cryptoadvance.specter==1.3.0) (3.4)
Requirement already satisfied: pysocks==1.7.1 in /home/specter/.env/lib/python3.9/site-packages/PySocks-1.7.1-py3.9.egg (from cryptoadvance.specter==1.3.0) (1.7.1)
Requirement already satisfied: python-dotenv==0.13.0 in /home/specter/.env/lib/python3.9/site-packages/python_dotenv-0.13.0-py3.9.egg (from cryptoadvance.specter==1.3.0) (0.13.0)
Requirement already satisfied: requests==2.25.0 in /home/specter/.env/lib/python3.9/site-packages/requests-2.25.0-py3.9.egg (from cryptoadvance.specter==1.3.0) (2.25.0)
Requirement already satisfied: semver==2.10.2 in /home/specter/.env/lib/python3.9/site-packages/semver-2.10.2-py3.9.egg (from cryptoadvance.specter==1.3.0) (2.10.2)
Requirement already satisfied: six==1.12.0 in /home/specter/.env/lib/python3.9/site-packages/six-1.12.0-py3.9.egg (from cryptoadvance.specter==1.3.0) (1.12.0)
Requirement already satisfied: stem==1.8.0 in /home/specter/.env/lib/python3.9/site-packages/stem-1.8.0-py3.9.egg (from cryptoadvance.specter==1.3.0) (1.8.0)
Requirement already satisfied: typing-extensions==3.7.4.3 in /home/specter/.env/lib/python3.9/site-packages/typing_extensions-3.7.4.3-py3.9.egg (from cryptoadvance.specter==1.3.0) (3.7.4.3)
Requirement already satisfied: urllib3==1.25.10 in /home/specter/.env/lib/python3.9/site-packages/urllib3-1.25.10-py3.9.egg (from cryptoadvance.specter==1.3.0) (1.25.10)
Requirement already satisfied: werkzeug==1.0.1 in /home/specter/.env/lib/python3.9/site-packages/Werkzeug-1.0.1-py3.9.egg (from cryptoadvance.specter==1.3.0) (1.0.1)
Requirement already satisfied: wtforms==2.3.3 in /home/specter/.env/lib/python3.9/site-packages/WTForms-2.3.3-py3.9.egg (from cryptoadvance.specter==1.3.0) (2.3.3)
Requirement already satisfied: zipp==3.3.0 in /home/specter/.env/lib/python3.9/site-packages/zipp-3.3.0-py3.9.egg (from cryptoadvance.specter==1.3.0) (3.3.0)
**ERROR: Package 'cryptoadvance-specter' requires a different Python: 3.9.2 not in '<3.9,>=3.6'**

kristapsk, 2022-09-08:
> ERROR: Package 'cryptoadvance-specter' requires a different Python: 3.9.2 not in '<3.9,>=3.6'

Specter Desktop required Python 3.6, 3.7 and 3.8. It [changed about two weeks ago](https://github.com/cryptoadvance/specter-desktop/pull/1847), now supported ones are 3.7, 3.8 and 3.9. You should just try again now!

kristapsk, 2022-09-15:
@fwittingham If this is still actual for you, try testing #1091.

fwittingham, 2022-09-20:
With the last version v1.12.0 I successfully installed Specter in RaspiBolt v3

PR #1091 (2022-09-21): Fixes for Specter desktop guide (mainly for RaspiBolt v3 + Specter version bump)
#### What / Why / How

Specter Desktop bonus guide wasn't updated since RaspiBolt v2 and wasn't working as is on v3. This is minimum changes I made to get it working on v3 + version bump (required for newer Python version support, among other things). Most breaking was migrating from Bitcoin RPC user/password to cookie auth in v3. Unfortunatelly, Specter does not allow to specify RPC cookie file location, only way is to link whole bitcoin datadir and then let it autoguess settings.

Not sure this is perfect, don't [... cut]
closed 2022-09-20 by fwittingham