raspibolt/raspibolt #1047: Bitcoind.service: Failed to parse PID from file

Stewie1786, 2022-07-04:
### Describe the issue
Dear Ladies & Gents,
I started following the guide to setup my first Raspibolt FullNode. My bitcoind.service gives me this error message:
**bitcoind.service: Failed to parse PID from file /run/bitcoind/bitcoind.pid: Invalid argument**
I do not know, how severe this is, since the service properly started syncing with the blockchain. Anyway, I would prefer to fix this before proceeding with further steps in the setup guide to avoid issues. Sorry for the noob question and thank you for advise.

### Location in guide
https://raspibolt.org/guide/bitcoin/bitcoin-client.html

### Screenshots
[image]

### Environment
 - Hardware platform: Raspberry Pi 4b (8GB)
 - Operating system: Raspberry Pi OS Lite 64bit
 - Version: Debian GNU/Linux 11 (bullseye)

### Additional context
Bitcoin debug.log: [](https://termbin.com/ccn1)
Syslog: [](https://termbin.com/ccn1)
Bitcoin.conf [](https://termbin.com/jh0s)

fwittingham, 2022-07-04:
See:
[https://github.com/rootzoll/raspiblitz/issues/1901](url)

Stewie1786, 2022-07-04:
> See: [https://github.com/rootzoll/raspiblitz/issues/1901](url)
I read this and tried the first possible solution to solve the issue by stopping bitcoind.service and deleting the PID file. Did not work.
The error message seems a bit different to me than the one mentioned in the thread. "Failed to parse file: invalid argument" is not the same as "File not found". Or is this the same in this case?

 The possible solutions shown in the thread are not really clear to me. Should I...
1.) Delete the PID file from service. --> What is the purpose of the file? Which functionalites would I lose?
2.) Change the PID file path --> Change the path in bitcoind.service, since its not correct?

Later on in the thread it becomes even more confusing. Please elaborate, Thanks!

closed 2022-09-09 by github-actions[bot]