raspibolt/raspibolt #1057: [BUG] `poetry install` in CLN bonus guide fails for me

kristapsk, 2022-08-02:
### Describe the bug
```
$ poetry install
Creating virtualenv cln-meta-project-TpDLTRw9-py3.8 in /home/cln/.cache/pypoetry/virtualenvs

  ModuleNotFoundError

  No module named 'virtualenv.seed.via_app_data'

  at <frozen importlib._bootstrap>:973 in _find_and_load_unlocked
```

### Location in guide
https://raspibolt.org/guide/bonus/lightning/cln.html

### Expected behavior
Command should not fail.

### Environment
 - Hardware platform: Odroid HC4
 - Operating system: Ubuntu Server
 - Version: Ubuntu 20.04.4 LTS

### Additional context
Any dependencies I may be missing? `virtualenv` itself is installed.

Also noticed in [most recent CLN install guide](https://github.com/ElementsProject/lightning/blob/af2b863b4a3a5e56c10ba20a1af5ff969379543c/doc/INSTALL.md), it now has `poetry run make` instead of `make`.

blckbx, 2022-08-02:
Thanks for the hints! I will add the new installation commands to #1056

Regarding virtualenv: https://stackoverflow.com/questions/63491221/modulenotfounderror-no-module-named-virtualenv-seed-embed-via-app-data-when-i Could this be some kind of version clash of virtualenv?

blckbx, 2022-08-03:
Updated my amd64 installation with these new poetry instructions just fine. Not sure if it's some configuration issue

kristapsk, 2022-08-03:
I also don't have issues with `poetry install` on amd64 Gentoo, but I do have on that specific arm64 Ubuntu. Will try to figure out what's wrong there.

ericsupreme, 2022-09-06:
same here. attached is the error message i get one running poetry install
[image]

ericsupreme, 2022-09-15:
> I also don't have issues with `poetry install` on amd64 Gentoo, but I do have on that specific arm64 Ubuntu. Will try to figure out what's wrong there.

were you able to figure anything out??

kristapsk, 2022-09-15:
> were you able to figure anything out??

No, not yet.

kristapsk, 2022-09-15:
@ericsupreme But your error looks different, could be something else.

ericsupreme, 2022-09-15:
> @ericsupreme But your error looks different, could be something else.

yes definitely but i caould at least start somewhere, i'm stumped lol

PR #1094 (2022-09-20): CLN: fix building and installation instructions
#### What

Reverts instructions to build from source. 
According to CLN docs: poetry is used for development environments, pip3 and make for production environments. 

#### Why

Fixes #1057
poetry didn't work on arm64 systems causing error on `poetry install`. On amd64 systems it did work tho.

#### How

Replacing poetry installation steps with pip, configure and make commands. 

#### Scope

- [ ] significant change to core configuration
- [ ] independent bonus guide
- [x] simple bug fix

Fixes #1057
closed 2022-09-20 by kristapsk