raspibolt/raspibolt #1059: Sparrow will not connect to raspibolt locally (Connection refused)

uskrusader, 2022-08-02:
My Sparrow and Electrum wallets were connected to my raspibolt via .onion.  I never got the local connection to work without TOR to connect.  Now my TOR is not working for some reason. (another issue)

Strangely, my wasabi wallet seems to connect just fine (because not electrs?)

I've tried the following troubleshooting in the guide:
-Computer is connected to non-guest wifi with password
-I've emptied the certs in:    On Linux: ~/.sparrow/certs
-I also restarted electrs (because my wasabi wallet works)

No luck, please help

https://raspibolt.org/guide/bitcoin/desktop-wallet.html

[image]

compane, 2022-08-04:
Set the log_filters = "DEBUG" in electrs.conf and restart the service "sudo systemctl restart electrs". Then run sudo journalctl -f -u electrs on your raspibolt while testing the Sparrow connection. 

Have you checked the UFW rules? Incoming connections from local lan on port 50002 should be allowed.

uskrusader, 2022-08-04:
ok, thanks, 
I set log_filters = "DEBUG" in electrs.conf and restarted the service "sudo systemctl restart electrs"  I ran sudo journalctl -f -u electrs .....it ran, but Sparrow continues to immediately refuse the connection.

I'm not sure what you mean by "UFW rules", my brain might need refreshing, I followed the guide months ago

compane, 2022-08-04:
The UFW rules are looking right.

I assume that the electrs daemon was running when you tried `electrs@raspibolt:~ $ electrs --conf /data/electrs/electrs.conf`.

The second screenshot shows that electrs in running. Please run `sudo journalctl -f -u electrs | grep :server` and try to test the connection. There should be an log entry containing "Sparrow". But I guess the connection will be refused again.

Is nginx running? `sudo systemctl status nginx`

uskrusader, 2022-08-04:
whoa, did we find the culprit?  Is nginx something i can just restart?

[image]

compane, 2022-08-04:
Restarting nginx won't fix your problem. It seems your "mempool.space" nginx config is faulty. You should check the guide (https://raspibolt.org/guide/bonus/bitcoin/mempool.html) again.

uskrusader, 2022-08-04:
hmmm? so weird, are u sure.  So I did this:  `sudo systemctl start nginx`  and sparrow now seems to be connecting.
Am I missing something?....thanks again for the help!

[image]

uskrusader, 2022-08-04:
I I just did that mempool guide, so ill recheck everything.
In the meantime, I should change that "debug" back to "info" right? in the electrs.conf?

VajraOfIndra, 2022-08-06:
> hmmm? so weird, are u sure?. So I did this: `sudo systemctl start nginx` and sparrow now seems to be connecting. Am I missing something?....thanks again for the help!

It's a known issue with the Mempool guide and how it interacts with nginx... it's a real pain. Usually restarting nginx fixes the issue. We hope to fix the issue definitively very soon, now that some code changes have been made in Mempool.
Glad you found how to fix your issue.

uskrusader, 2022-08-10:
I would close this issue, but I'm still having problems connecting my Sparrow wallet to my node.  Sparrow only connects via .onion about 20% of the time.  My wasabi wallet still seems to connect fine with it's own TOR.  Are TOR issues affecting other people with Sparrow?  Everytime I run systemctl status tor@default I get alot of weird responses from both my raspibolt and my remote's own CL.

Whenever I test my connection in Sparrow, it always fails.... but then in the main wallet, it sometimes connects to my node with TOR and the proper block height.... randomly it seems.  Weird. So is it TOR?  Thanks!

uskrusader, 2022-09-13:
This just happened again, again it was nginx on my raspibolt that failed.  sudo stop nginx / sudo start nginx seems to have fixed my problem.

closed 2022-09-13 by uskrusader

VajraOfIndra, 2022-09-14:
> This just happened again, again it was nginx on my raspibolt that failed. sudo stop nginx / sudo start nginx seems to have fixed my problem.

Just my 2 sats, but I suggest uninstalling Mempool and not using it until we fixed the issue with it.  That's what I'm doing on my node :/