raspibolt/raspibolt #1065: [FEATURE REQ] LND remote static channel backups should be pushed via Tor

kristapsk, 2022-08-16:
### What & why

Currently RaspiBolt configures LND to work as .onion service only by default. But remote static channel pushes to github.com happen using clearnet. This way your real IP address ir revealed to github.com (Microsoft), together with your channel information.

### How

git pushes should happen using SOCKS5 localhost Tor proxy.

twofaktor, 2022-09-06:
How do you suggest this could be deployed? Which commands to change?

kristapsk, 2022-09-06:
Not sure yet. Either git can be configured to do this somehow or maybe `torsocks` could be used.

blckbx, 2022-09-11:
probably via
`git config --global http.proxy http://127.0.0.1:9050`
and I guess we have to be torified (TorBrowser) right from the start at the time of the github account and repo creation as well.

kristapsk, 2022-09-11:
> `git config --global http.proxy http://127.0.0.1:9050`

Don't think that will work, Tor provides SOCKS5 proxy, not a HTTP proxy.

blckbx, 2022-09-12:
https://stackoverflow.com/questions/15227130/using-a-socks-proxy-with-git-for-the-http-transport
despite the naming, socks5 should also work here (to be tested)
`git config --global http.proxy 'socks5://127.0.0.1:9050'`

kristapsk, 2022-09-14:
Ok, likely next problem, static channel backups is pushed to github via ssh not http. But [`core.sshCommand`](https://git-scm.com/docs/git-config#Documentation/git-config.txt-coresshCommand) setting seems to be the right one:

> **core.sshCommand**
> If this variable is set, `git fetch` and `git push` will use the specified command instead of `ssh` when they need to connect to a remote system. The command is in the same form as the `GIT_SSH_COMMAND` environment variable and is overridden when the environment variable is set.

PR #1089 (2022-09-15): Push remote LND static channel backups to GitHub via Tor
#### What

Push LND local channel remote backups to GitHub via Tor, not clearnet.

### Why

RaspiBolt configures LND to work as .onion service only by default. But without this change remote static channel pushes to github.com happen using clearnet. This way your real IP address ir revealed to github.com (Microsoft), together with your channel information.

#### How

Change git configuration for lnd user to use `torsocks ssh` instead of `ssh` to connect to GitHub.

#### Scope

- [X] significant change to core configuration
- [ ] independ [... cut]
closed 2022-09-15 by kristapsk