raspibolt/raspibolt #1067: [BUG] LND/LIT won't connect after a restart

pleb4868, 2022-08-22:
### Describe the bug
After setting the node for i2p connectivity and restarting bitcoind. lightning terminal doesn't start gives a 401 error.
the block explorer and joinmarket run fine

### Location in guide
[https://raspibolt.org/guide/lightning/lightning-client.html#run-lnd](url)
or
[https://raspibolt.org/guide/lightning/lightning-client.html#run-lnd](url)

### Expected behavior
lnd/lit starts working after "sudo systemctl start litd"

### Environment
 - Hardware platform: RPi4 8GB
 - Operating system: Raspberry Pi OS Lite (64-bit)
 - Version: Debian GNU/Linux 11 (bullseye)

### Additional context
lightning terminal log:

systemd[1]: Started Lightning Terminal Daemon.
[INF] LITD: LiT version: 0.7.0-alpha commit=v0.7.0-alpha
[INF] SESS: Checking for schema update: latest_version=0, db_version=0
[INF] LITD: Dialing lnd gRPC server at 127.0.0.1:10009
[DBG] GRPC: [core] parsed scheme: ""
[DBG] GRPC: [core] scheme "" not registered, fallback to default scheme
[DBG] GRPC: [core] ccResolverWrapper: sending update to cc: {[{127.0.0.1:10009  <nil> 0 <nil>}] <nil> <nil>}
[DBG] GRPC: [core] ClientConn switching balancer to "pick_first"
[DBG] GRPC: [core] Channel switches to new LB policy "pick_first"
[DBG] GRPC: [core] Subchannel Connectivity change to CONNECTING
[DBG] GRPC: [core] pickfirstBalancer: UpdateSubConnState: 0x400003a2c0, {CONNECTING <nil>}
[DBG] GRPC: [core] Subchannel picks a new address "127.0.0.1:10009" to connect
[DBG] GRPC: [core] Channel Connectivity change to CONNECTING
[DBG] GRPC: [core] parsed scheme: ""
[DBG] GRPC: [core] scheme "" not registered, fallback to default scheme
[DBG] GRPC: [core] ccResolverWrapper: sending update to cc: {[{127.0.0.1:10009  <nil> 0 <nil>}] <nil> <nil>}
[DBG] GRPC: [core] ClientConn switching balancer to "pick_first"
[DBG] GRPC: [core] Channel switches to new LB policy "pick_first"
[DBG] GRPC: [core] Subchannel Connectivity change to CONNECTING
[DBG] GRPC: [core] blockingPicker: the picked transport is not ready, loop back to repick
[INF] LITD: Listening for http_tls on: [::]:8443
[DBG] GRPC: [core] pickfirstBalancer: UpdateSubConnState: 0x400003a750, {CONNECTING <nil>}
[DBG] GRPC: [core] Subchannel picks a new address "127.0.0.1:10009" to connect
[DBG] GRPC: [core] Channel Connectivity change to CONNECTING
[DBG] GRPC: [core] Subchannel Connectivity change to READY
[DBG] GRPC: [core] pickfirstBalancer: UpdateSubConnState: 0x400003a2c0, {READY <nil>}
[DBG] GRPC: [core] Channel Connectivity change to READY
[DBG] GRPC: [core] Subchannel Connectivity change to READY
[DBG] GRPC: [core] pickfirstBalancer: UpdateSubConnState: 0x400003a750, {READY <nil>}
[DBG] GRPC: [core] Channel Connectivity change to READY
[DBG] GRPC: [core] Channel Connectivity change to SHUTDOWN
[DBG] GRPC: [core] Subchannel Connectivity change to SHUTDOWN
[INF] SGNL: Received shutdown request.
litd[402344]: error displaying startup info: error querying remote node : rpc error: code = Unknown desc = unable to get best block info: status code: 401, response: ""
systemd[1]: litd.service: Main process exited, code=exited, status=1/FAILURE
systemd[1]: litd.service: Failed with result 'exit-code'.

twofaktor, 2022-08-23:
Hello, are you sure that the problems appeared after setting the node for i2p connectivity?

Did you try stopping and disabling i2pd service, commenting on the lines bitcoin.conf, checking to restart all services again?

pleb4868, 2022-08-23:
Hi @twofaktor 
I'm not sure if the i2p connectivity is to blame but the bug started after adding the lines and restarting bitcoind.

I stopped the i2pd service and still got the same error.

I commented out the i2p lines from bitcoin.conf and restarted the bitcoind.service and still got the same result.

It seems any lncli command results in the error:
rpc error: code = Unknown desc = unable to get best block info: status code: 401, response: ""

* as an aside before commenting out the i2p lines from bitcoin.conf: when running the bitcoin-cli -netinfo command the i2p address has port 0 which i assume is invalid.
abc...xyz.b32.i2p       port      0    score      4

pleb4868, 2022-08-28:
I seems this issue was related to #1052 
I upgraded my LND, restarted it and then 401 errors stopped appearing and the node is running fine...

pleb4868, 2022-09-05:
been running fine for a while now, closing

closed 2022-09-05 by pleb4868