raspibolt/raspibolt #1080: I2P running for a few weeks, no connected peers yet..?

VajraOfIndra, 2022-08-31:
### Describe the issue
I followed this guide to run I2P: https://raspibolt.org/guide/bonus/bitcoin/i2p.html

After a few weeks, still no peers. I suspect there is an issue somewhere...
[image]

### Location in guide
[Link to the corresponding page and section on <https://raspibolt.org>](https://raspibolt.org/guide/bonus/bitcoin/i2p.html)

### Screenshots
I2P service is active:
[image]

### Environment
 - Hardware platform: Raspberry Pi 4 4GB
 - Operating system: RasPiOS Lite 64bit
 - Version: Debian GNU/Linux 11 (bullseye)

### Additional context
Logs output:

> admin@raspibolt:~ $ sudo tail -f /var/log/i2pd/i2pd.log
> 13:00:19@245/warn - SSU2: Session with 37.204.86.74:9718 was not established after 5 seconds
> 13:00:43@329/error - Garlic: Failed to decrypt message
> 13:00:46@726/warn - NTCP2: SessionCreated read error: End of file
> 13:00:47@352/warn - NetDb: Destination xjboEpb8uwkMPcy2uzu8I63W-oznn91M0nD~6ofswXQ= is requested already
> 13:01:29@329/error - Garlic: Failed to decrypt message
> 13:02:32@726/warn - NTCP2: SessionCreated read error: End of file
> 13:03:06@846/warn - Tunnels: Test of tunnel 2341950696 failed
> 13:03:11@846/warn - Tunnels: Test of tunnel 2819269901 failed
> 13:03:57@751/warn - Transports: Session to peer ZEnMpbQzgZwAOGdtvS5oKltYYKzfN35~G8kWbT0KeRo= has not been created in 15 seconds
> 13:04:06@846/warn - Tunnels: Test of tunnel 1738627153 failed
> 13:04:36@846/warn - Tunnels: Test of tunnel 508700813 failed
> 13:04:46@846/warn - Tunnels: Test of tunnel 224718362 failed
> 13:04:47@726/warn - NTCP2: SessionCreated read error: End of file
> 13:05:06@726/warn - NTCP2: Receive length read error: End of file
> 13:05:13@329/error - Garlic: Failed to decrypt message
> 13:05:16@846/warn - Tunnels: Test of tunnel 3144601582 failed
> 13:05:18@726/warn - NTCP2: SessionCreated read error: End of file
> 13:05:18@352/warn - NetDb: Destination henVNN0cDWwNMcvrDZnnnHDf5e~I~57iVpQpiHEhOFg= is requested already
> 13:05:29@329/error - Garlic: Failed to decrypt message
> 13:05:41@846/warn - Tunnels: Test of tunnel 369296486 failed
> 13:06:12@726/warn - NTCP2: SessionCreated read error: End of file
> 13:06:26@846/warn - Profiling: No profile yet for 1dMtr5HyvIIUXtxSebw1tV1-CyMrvAg-ipSGTN-Ft7E=
> 13:07:19@726/warn - NTCP2: Receive length read error: End of file
> 13:07:28@329/error - Garlic: Failed to decrypt message
> 13:07:31@846/warn - Tunnels: Test of tunnel 1593075819 failed

kristapsk, 2022-08-31:
I would suggest to try manually connect to some I2P peers listed here - https://github.com/bitcoin/bitcoin/blob/master/contrib/seeds/nodes_main.txt (`bitcoin-cli addnode <address> add`). Also, does `bitcoin-cli -netinfo` shows your I2P address?

VajraOfIndra, 2022-08-31:
> I would suggest to try manually connect to some I2P peers listed here - https://github.com/bitcoin/bitcoin/blob/master/contrib/seeds/nodes_main.txt (`bitcoin-cli addnode <address> add`). 

Thanks! Done. I'll monitor the peer count now that I have one..

> Also, does `bitcoin-cli -netinfo` shows your I2P address?

Yes, with port 0 and score 4

twofaktor, 2022-09-06:
Yeah i2p is running with port 0 by the way: I have 3 incoming peers, i2p is growing little by little 🚀 all cleared? 

[image]

Can this support ticket be closed?

closed 2022-09-06 by VajraOfIndra

soc7, 2024-11-30:
usually restarting the service fixes it.