raspibolt/raspibolt #1081: [BUG] 403 Forbidden on mempool install

visskiss, 2022-08-31:
### Describe the bug
Followed instructions for install but used latest mempool 2.4.0 and latest node v18

Following error in mempool-error.log

[error] 23456#23456: *51 directory index of "/var/www/mempool/browser/en-US//" is forbidden, client: fe80::cf5:999f:8bfe:8ff, server: _, request: "GET / HTTP/1.1",

### Location in guide
[Link to the corresponding page and section on <https://raspibolt.org>](https://raspibolt.org/guide/bonus/bitcoin/mempool.html)

### Expected behavior
Not a 403 forbidden

### Environment
raspberry pi

VajraOfIndra, 2022-09-02:
Did you use https (with the s at end) in the URL in your browser..?

visskiss, 2022-09-02:
Node 16 is required and I was using node 18. Closing.

closed 2022-09-02 by visskiss