raspibolt/raspibolt #1082: ThunderHub doesn't start after upgrade.

Ldngzero, 2022-09-01:
### Describe the issue
After upgrading ThunderHub, the service does not start anymore.

### Location in guide
[Link to the corresponding page and section on <https://raspibolt.org>](https://raspibolt.org/guide/bonus/lightning/thunderhub.html#upgrade)

### Screenshots
admin@raspibolt:~ $ sudo journalctl -f -u thunderhub
-- Journal begins at Fri 2022-07-08 11:17:07 BST. --
Sep 01 13:10:32 raspibolt systemd[1]: Stopped Thunderhub.
Sep 01 13:10:32 raspibolt systemd[1]: Started Thunderhub.
Sep 01 13:10:32 raspibolt npm[64075]: /usr/bin/env: ‘node’: No such file or directory
Sep 01 13:10:32 raspibolt systemd[1]: thunderhub.service: Main process exited, code=exited, status=127/n/a
Sep 01 13:10:32 raspibolt systemd[1]: thunderhub.service: Failed with result 'exit-code'.
Sep 01 13:11:01 raspibolt systemd[1]: Stopped Thunderhub.

admin@raspibolt:~ $ node -v
v18.8.0

### Environment
Raspberry Pi OS 64bit, Debian 11

### Additional context
There seems to be an issue with Node.js. The guide is suggesting version 16. I have v18 installed.
What could be the best way to solve this issue with ThunderHub and Node.js?

twofaktor, 2022-09-02:
Install the latest LTS Nodejs version 16.x, maybe this could be a solution:

```
$ sudo apt --purge autoremove nodejs"
$ curl -fsSL https://deb.nodesource.com/setup_16.x | sudo -E bash -"
$ sudo apt install nodejs"
```

and try to start thunderhub again

Ldngzero, 2022-09-02:
Thank you, this worked! 

PS There were some warnings though:

admin@raspibolt:~ $ sudo apt --purge autoremove nodejs
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages will be REMOVED:
  fontconfig* libaacs0* libaom0* libass9* libasyncns0* libavc1394-0* libavcodec58* libavdevice58* libavfilter7* libavformat58* libavutil56* libbdplus0* libblas3* libbluray2* libbs2b0*
  libcaca0* libcairo-gobject2* libcairo2* libcdio-cdda2* libcdio-paranoia2* libcdio19* libchromaprint1* libcodec2-0.9* libdatrie1* libdav1d4* libdc1394-25* libdrm-amdgpu1*
  libdrm-nouveau2* libdrm-radeon1* libepoxy0* libfftw3-double3* libflac8* libflite1* libgbm1* libgdk-pixbuf-2.0-0* libgdk-pixbuf2.0-bin* libgdk-pixbuf2.0-common* libgfortran5* libgl1*
  libgl1-mesa-dri* libglapi-mesa* libglvnd0* libglx-mesa0* libglx0* libgme0* libgraphite2-3* libgsm1* libharfbuzz0b* libiec61883-0* libjack-jackd2-0* liblapack3* liblilv-0-0* libmp3lame0*
  libmpg123-0* libmysofa1* libnorm1* libnuma1* libogg0* libopenal-data* libopenal1* libopenjp2-7* libopenmpt0* libopus0* libpango-1.0-0* libpangocairo-1.0-0* libpangoft2-1.0-0*
  libpgm-5.3-0* libpixman-1-0* libpocketsphinx3* libpostproc55* libpulse0* librabbitmq4* libraw1394-11* librsvg2-2* librsvg2-common* librubberband2* libsdl2-2.0-0* libsensors-config*
  libsensors5* libserd-0-0* libshine3* libsnappy1v5* libsndfile1* libsndio7.0* libsodium23* libsord-0-0* libsoxr0* libspeex1* libsphinxbase3* libsratom-0-0* libsrt1.4-gnutls*
  libssh-gcrypt-4* libswresample3* libswscale5* libthai-data* libthai0* libtheora0* libtwolame0* libudfread0* libva-drm2* libva-x11-2* libva2* libvdpau-va-gl1* libvdpau1* libvidstab1.1*
  libvorbis0a* libvorbisenc2* libvorbisfile3* libvpx6* libvulkan1* libwavpack1* libwayland-client0* libwayland-cursor0* libwayland-egl1* libwayland-server0* libwebpmux3* libx11-xcb1*
  libx264-160* libx265-192* libxcb-dri2-0* libxcb-dri3-0* libxcb-glx0* libxcb-present0* libxcb-randr0* libxcb-render0* libxcb-shape0* libxcb-shm0* libxcb-sync1* libxcb-xfixes0*
  libxcursor1* libxdamage1* libxfixes3* libxi6* libxinerama1* libxkbcommon0* libxrandr2* libxrender1* libxshmfence1* libxss1* libxv1* libxvidcore4* libxxf86vm1* libzmq5* libzvbi-common*
  libzvbi0* mesa-va-drivers* mesa-vdpau-drivers* mesa-vulkan-drivers* nodejs* ocl-icd-libopencl1* pocketsphinx-en-us* va-driver-all* vdpau-driver-all* x11-common*
0 upgraded, 0 newly installed, 154 to remove and 7 not upgraded.
After this operation, 407 MB disk space will be freed.
Do you want to continue? [Y/n] Y
(Reading database ... 50716 files and directories currently installed.)
Removing libavdevice58:arm64 (7:4.3.4-0+deb11u1+rpt3) ...
Removing libavfilter7:arm64 (7:4.3.4-0+deb11u1+rpt3) ...
Removing libavformat58:arm64 (7:4.3.4-0+deb11u1+rpt3) ...
Removing libchromaprint1:arm64 (1.5.0-2) ...
Removing libavcodec58:arm64 (7:4.3.4-0+deb11u1+rpt3) ...
Removing librsvg2-common:arm64 (2.50.3+dfsg-1) ...
Removing librsvg2-2:arm64 (2.50.3+dfsg-1) ...
Removing libpangocairo-1.0-0:arm64 (1.46.2-3) ...
Removing libpangoft2-1.0-0:arm64 (1.46.2-3) ...
Removing libpango-1.0-0:arm64 (1.46.2-3) ...
Removing fontconfig (2.13.1-4.2) ...
Removing libaacs0:arm64 (0.9.0-2) ...
Removing libaom0:arm64 (1.0.0.errata1-3) ...
Removing libass9:arm64 (1:0.15.0-2) ...
Removing libpocketsphinx3:arm64 (0.8+5prealpha+1-13) ...
Removing libsphinxbase3:arm64 (0.8+5prealpha+1-12) ...
Removing libsdl2-2.0-0:arm64 (2.0.14+dfsg2-3+deb11u1) ...
Removing libpulse0:arm64 (14.2-2+rpt1) ...
Removing libasyncns0:arm64 (0.8-6+b2) ...
Removing libavc1394-0:arm64 (0.5.4-5) ...
Removing libswscale5:arm64 (7:4.3.4-0+deb11u1+rpt3) ...
Removing libswresample3:arm64 (7:4.3.4-0+deb11u1+rpt3) ...
Removing libbdplus0:arm64 (0.1.2-3) ...
Removing liblapack3:arm64 (3.9.0-3) ...
Removing libblas3:arm64 (3.9.0-3) ...
Removing libbluray2:arm64 (1:1.2.1-4+deb11u1) ...
Removing libbs2b0:arm64 (3.1.0+dfsg-2.2+b1) ...
Removing libcaca0:arm64 (0.99.beta19-2.2) ...
Removing libcairo-gobject2:arm64 (1.16.0-5+rpt1) ...
Removing libtheora0:arm64 (1.1.1+dfsg.1-15) ...
Removing libcairo2:arm64 (1.16.0-5+rpt1) ...
Removing libcdio-paranoia2:arm64 (10.2+2.0.0-1+b2) ...
Removing libcdio-cdda2:arm64 (10.2+2.0.0-1+b2) ...
Removing libcdio19:arm64 (2.1.0-2) ...
Removing libcodec2-0.9:arm64 (0.9.2-4) ...
Removing libthai0:arm64 (0.1.28-3) ...
Removing libdatrie1:arm64 (0.2.13-1) ...
Removing libdav1d4:arm64 (0.7.1-3) ...
Removing libdc1394-25:arm64 (2.2.6-3) ...
Removing mesa-vulkan-drivers:arm64 (20.3.5-1+rpt4+rpi1) ...
Removing vdpau-driver-all:arm64 (1.4-3) ...
Removing mesa-vdpau-drivers:arm64 (20.3.5-1+rpt4+rpi1) ...
Removing va-driver-all:arm64 (2.10.0-1) ...
Removing mesa-va-drivers:arm64 (20.3.5-1+rpt4+rpi1) ...
Removing libvdpau-va-gl1:arm64 (0.4.2-1+b1) ...
[... 359 lines cut ...]
Preparing to unpack .../nodejs_17.9.0-deb-1nodesource1_arm64.deb ...
Unpacking nodejs (17.9.0-deb-1nodesource1) ...
Setting up nodejs (17.9.0-deb-1nodesource1) ...
Processing triggers for man-db (2.9.4-2) ...
admin@raspibolt:~ $ node -v
v18.8.0
admin@raspibolt:~ $ sudo systemctl stop thunderhub
admin@raspibolt:~ $ sudo systemctl start thunderhub
admin@raspibolt:~ $ sudo journalctl -f -u thunderhub
-- Journal begins at Fri 2022-07-08 11:17:07 BST. --
Sep 02 12:14:52 raspibolt systemd[1]: Stopped Thunderhub.
Sep 02 12:14:52 raspibolt systemd[1]: Started Thunderhub.
[... cut ...]
Sep 02 12:18:30 raspibolt npm[71445]: (node:71445) [DEP0123] DeprecationWarning: Setting the TLS ServerName to an IP address is not permitted by RFC 6066. This will be ignored in a future version.
Sep 02 12:18:30 raspibolt npm[71445]: (Use `node --trace-deprecation ...` to show where the warning was created)
Sep 02 12:20:26 raspibolt npm[71445]: Warning: For production Image Optimization with Next.js, the optional 'sharp' package is strongly recommended. Run 'yarn add sharp', and Next.js will use it automatically for Image Optimization.
Sep 02 12:20:26 raspibolt npm[71445]: Read more: https://nextjs.org/docs/messages/sharp-missing-in-production

closed 2022-09-02 by Ldngzero