raspibolt/raspibolt #1086: [FEATURE REQ] Add bandwith limiting guidance to I2P guide

kristapsk, 2022-09-06:
### What & why

Some people running RaspiBolt may have limited bandwidth available, so they may want to limit I2P routing happening.

### How

There is similar PR now open for Bitcoin Core - https://github.com/bitcoin/bitcoin/pull/25993.

twofaktor, 2022-09-07:
What values do you suggest to put in bitcoin.conf?

The next example is to limit total I2P traffic to 256KB/s, and share 50% of this limit for a maximum of 20 transit tunnels:
```
bandwidth = 256
share = 50

[limits]
transittunnels = 20
```
Should we put `notransit=true` instead?
`If you prefer not to relay any public I2P traffic and only permit I2P traffic from programs which are connecting via the SAM proxy, e.g. Bitcoin Core, you can set the notransit option to true.`

kristapsk, 2022-09-07:
> What values do you suggest to put in bitcoin.conf?

Guess you meant i2pd.conf?

I don't have strong opinion on default values, don't have bandwidth issues myself. Maybe should just use the sames as are merged in Bitcoin Core docs. Most important part is describing these configuration options, so that user can decide for himself.

> Should we put `notransit=true` param instead?

Would be good to describe it, so, again, user can make his own decision.

If user has no bandwidth issues and can afford it, it's good to not have this, so then his node benefits I2P network itself, even without Bitcoin part.

VajraOfIndra, 2022-09-08:
> Most important part is describing these configuration options, so that user can decide for himself.

Sounds like a good approach. With maybe a pointer on how to check bandwith usage from the node? I don't know how to do that myself so will look it up on the internet but it could be nice to have it in the guide. Any recommendation on bets way to do this on a Pi?

---

Anyone interested to tackle and write this pull request?

twofaktor, 2022-09-10:
> > Most important part is describing these configuration options so that the user can decide for himself.
> 
> Sounds like a good approach. With maybe a pointer on how to check bandwith usage from the node? I don't know how to do that myself so will look it up on the internet but it could be nice to have it in the guide. Any recommendation on best way to do this on a Pi?
> 
> Anyone interested to tackle and write this pull request?

> > Most important part is describing these configuration options, so that user can decide for himself.
> 
> Sounds like a good approach. With maybe a pointer on how to check bandwith usage from the node? I don't know how to do that myself so will look it up on the internet but it could be nice to have it in the guide. Any recommendation on best way to do this on a Pi?
> 
> Anyone interested to tackle and write this pull request?

I could, but I will need help to determine the correct values of i2p.conf. Leave here a screenshot of the complete bandwidth configuration in i2p.conf that I found as a reference

[image]

VajraOfIndra, 2022-09-17:
> The next example is to limit total I2P traffic to 256KB/s, and share 50% of this limit for a maximum of 20 transit tunnels:
> ```
> bandwidth = 256
> share = 50
> 
> [limits]
> transittunnels = 20
> ```

@kristapsk @twofaktor 

What does it mean that I2P will "share 50%" of the 256KB/s limit..?

Also, what is the impact of the number of tunnels on I2P, our bandwith or the node..? Since there is a maximum limit to th ebandwith that I2P can use, what;s the effect of the number of tunnels?

Finally, what would you do to select the right values for you, step by step? (e.g. testing one's own internet speed on the node? checking with the ISP for any bandwith and speed limit? etc etc)

I went for notransit=true for now for the sake of simplicity but I'd like to learn more to be able to support I2P while being confident I'm ruining my home internet! :)

openoms, 2022-09-25:
It seems that if you don't specify and forward the port there is no bandwith used for relaying by default: https://github.com/seth586/guides/blob/master/FreeNAS/bitcoin/freenas_3_tor.md#optional-relay

So setting up a relay would be an optional extra step.

twofaktor, 2022-09-27:
PR opened in #1104

jonatack, 2022-12-29:
See https://github.com/bitcoin/bitcoin/issues/26754 for why this may not be a good idea, namely:

*Also in that doc, in the bandwidth section at the bottom, please encourage people to share as much bandwidth and transit tunnels as they can, to increase their anonymity with the cover traffic, and so that bitcoin users contribute more to the i2p network than they consume.*

*If a popular application uses more network resources than it contributes, it has the potential to take down the entire network.*

zzzi2p, 2022-12-29:
> See [bitcoin/bitcoin#26754](https://github.com/bitcoin/bitcoin/issues/26754) for why this may not be a good idea, namely:
> 
> _Also in that doc, in the bandwidth section at the bottom, please encourage people to share as much bandwidth and transit tunnels as they can, to increase their anonymity with the cover traffic, and so that bitcoin users contribute more to the i2p network than they consume._
> 
> _If a popular application uses more network resources than it contributes, it has the potential to take down the entire network._

+++

"transittunnels = 20" is an exceptionally low number for all but the lamest of raspberry pi's, and could mislead people into thinking it's a recommendation.  Most routers will do fine at self-limiting anyway. "transittunnels = xxx" or "transittunnels = 9999" might be better and make more clear it's an example.

kristapsk, 2023-01-05:
@zzzi2p But it does some transit traffic even when no incoming ports for I2P are open in firewall?

zzzi2p, 2023-01-05:
yes, but probably not a lot. Please contact i2pd project for further questions and support.

kristapsk, 2023-01-05:
After comments above, I don't think anymore this is a good idea, would like to close if there are no objections.

closed 2023-01-05 by kristapsk