raspibolt/raspibolt #1099: Balance of Satoshis - Circular Rebalancing Typo?

dunwright, 2022-09-20:
### Describe the bug
multiple instances of bos command syntax ends with:
--out [NODE_**A**_PUBKEY]

### Location in guide
https://raspibolt.org/guide/bonus/lightning/balance-of-satoshis.html#circular-rebalancing

### Expected behavior
Shouldn't that syntax end with?:
--out [NODE_**B**_PUBKEY]

### Screenshots
[image]

### Environment
N/A

### Additional context
Example command syntax provided using Bitrefill node as node 'B' ends with --out using Bitrefill's pubkey.

VajraOfIndra, 2022-09-21:
Well spotted, thanks a lot! Can't believe I spent all that time explaining how the rebalancing works and still screwing up the actual command 😅 

I made a fix with #1100 .. should be merged soon! (this issue will close automatically, once that pull request will be merged)

VajraOfIndra, 2022-09-21:
> No sweat... documentation is complicated (especially keeping it up to date with changes in subcomponent versions). 

Yes, I get that, I feel we're always behind haha

> This project has me excited about technology again...

Nice to hear this ❤️ 
Me it got me learning about Linux and FOSS, just for that I'm thankful! :)

> I work for a large global cybersecurity manufacturer

Sounds pretty cool! I hope you don't mind me asking but if you can see ways we could harden the security of the whole setup, we'd love to hear about them!

Thanks for your help on this issue anyways, much appreciated.

PR #1100 (2022-10-08): Fixes typo in rebalancing command + small rewording of explanations
#### What

There is a typo in the rebalancing commands, the `--out` flag should be node B, not node A, as pointed out in #1099.
This PR contains:
- a fix for the typo: `NODE_A_PUBKEY `-> `NODE_B_PUBKEY`
- a rewording of the command introductory sentence to make it clearer  

#### Scope

- [ ] significant change to core configuration
- [ ] independent bonus guide
- [x] simple bug fix

Fixes #1099 

#### Test & maintenance

- Check that the typo has been fixed everywhere it appears.
- Check that the rewording of the instructions [... cut]
closed 2022-10-08 by VajraOfIndra