raspibolt/raspibolt #1106: same here. attached is the error message i get when running poetry install

ericsupreme, 2022-09-28:
same here. attached is the error message i get when running poetry install
[image]
i'm running a rasp pi 400 ( i know!! lol)

_Originally posted by @ericsupreme in https://github.com/raspibolt/raspibolt/issues/1057#issuecomment-1238344061_

kristapsk, 2022-09-28:
There is no `poetry install` in updated CLN instructions anymore. https://raspibolt.org/guide/bonus/lightning/cln.html

ericsupreme, 2022-09-28:
ahhhh

> There is no `poetry install` in updated CLN instructions anymore. https://raspibolt.org/guide/bonus/lightning/cln.html

ahhhh ty very much for the speedy reply. i will give it a whirl!!

closed 2022-09-28 by ericsupreme