raspibolt/raspibolt #1112: Electrum client / couldn't launch iface

ARKET2022, 2022-10-01:
### Describe the issue
A clear and concise description of where you're stuck in the guide. What are you trying to do, and how is it not working?

### Location in guide
Link to the corresponding page and section on <https://raspibolt.org>

### Screenshots
If applicable, add screenshots to help explain your problem.

### Environment
 - Hardware platform: [e.g. Raspberry Pi 4, Odroid, ... ]
 - Operating system: [e.g, Raspberry Pi OS 64bit, Debian 11, ... ]
 - Version: [ output of `lsb_release -d` ]

### Additional context
Add any other context about the problem here.

ARKET2022, 2022-10-01:
Hi ALL, I installed the raspibolt v3 step by step with the guide.
I tried to connet Electrum Client with the ElectrumSrv but I receive this error, red light on electrum client :
I/i | interface.[xxmyaddressxx.onion:50002] | disconnecting due to: CancelledError()
I/n | network | couldn't launch iface xxmyaddressxx.onion:50002:t -- CancelledError()
I/n | network | connecting to xxmyaddressxx.onion:50002:t as new interface

./electrum-4.3.2-x86_64.AppImage -1 -s myelectrumaddress.onion:50002:t -p socks5:127.0.0.1:9050 --rpcuser myuser --rpcpassword mypassword -v

Alle seems is working on raspibolt node. I tried to telnet the port 50002 and it is open.
On the PC tor is up and running. I used netstat, 9050 is listen on 127.0.0.1.
Please could you hep me? Many thanks.

VajraOfIndra, 2022-10-01:
Shouldn't  `myelectrumaddress.onion:50002:t` be with an `s` instead of a `t` like  `myelectrumaddress.onion:50002:t` ? Try to change this maybe.

Also, if you already had Electrum running on your laptop, you might need to delete a file in your local Electrum installation folder (the one with your node IP as title, I forgot in which folder it is)

closed 2022-12-08 by github-actions[bot]