raspibolt/raspibolt #1125: LNDG is not shown in browser with 502 bad Gateway.

ohohora, 2022-10-14:
I followed latest updated Lndg guide_10/13 (there is a link below). Before installing uwsgi, it worked on http://raspibolt.local:8889 but After setting up uwsgi, nginx, etc (web server configuration part), it no longer works on https://raspibolt.local:8889.
I restarted uwsgi, nginx but didn't work, and nginx error log file (/var/log/nginx/error.log) is as following :

2022/10/14 03:37:16 [crit] 990#990: *202 connect() to unix:///home/lndg/lndg/lndg.sock failed (13: Permission denied) while connecting to upstream, client: xx.xx.xx.xx, server: _, request: "GET / HTTP/1.1", upstream: "uwsgi://unix:///home/lndg/lndg/lndg.sock:", host: "raspibolt.local:8889"

lndg.sock file has been applied 660 well by the chmod, chown commands. I also restarted nginx, uwsgi but still don't work. 
I dont' know what's going on. Will appreciate with your help. 

### Location in guide
https://github.com/VajraOfIndra/RaspiBolt/blob/LNDg/guide/bonus/lightning/lndg.md

### Environment
 - Hardware platform: PC (amd64)
 - Operating system: ubuntu 22.04

Buttafuoco, 2022-10-16:
I have same issue with 400 code. The only way to connect is via HiddenService

ohohora, 2022-10-18:
Now It works on https://raspibolt.local:8889 by two ways each below. 

1. Adding user 'www-data' to group 'lndg' 
          $ sudo adduser --disabled-password --gecos "" lndg
          $ sudo adduser lndg lnd
          $ sudo adduser lndg www-data
          **$ sudo adduser www-data lndg**

or

2. Changing the user from 'www-data' to 'lndg' in /etc/nginx/nginx.conf 
          $ sudo nano /etc/nginx/nginx.conf 
            **#user www-data;
            user lndg;**
            worker_processes 1;

Maybe got rid of permission error(502 bad gateway) but these methods looks a bit weired. I doubt that if it will mess up nginx related services or other things. Need some advice, comments~!

VajraOfIndra, 2022-10-19:
> Now It works on https://raspibolt.local:8889 by two ways each below.
> 
>     1. Adding user 'www-data' to group 'lndg'
> [...]
> 
> or
>     2. Changing the user from 'www-data' to 'lndg' in /etc/nginx/nginx.conf
> [...]

Option 2 is not possible as this is the default nginx config file which is used for other apps (e.g. Mempool, Homer...)

Option 1 should be feasible I think.. The weird thing is that I have LNDg running on mine without having to have done that.. I'm not sure what's different on my setup vs the draft guide. 

I'm very new to nginx, uwsgi and Python stuff and it's so layered..! it's hard to keep track of how all the pieces work together..

ohohora, 2022-10-20:
Thanks~!!!  I'll run LNDg on option1.

VajraOfIndra, 2022-10-20:
> Like you said, option 1 looks feasible and thank you~!!!

Sounds good if it works for you for now but it shouldn't be necessary. I can access LNDg via https without this additional step for example (but don't know why yet! :). I'm reworking the guide to try to solve these permissions issues and find the best prtactices, keep an eye on the main PR.

ohohora, 2022-10-20:
I also it looks unnecessary step. I'll keep an eye on main PR~!!!

blckbx, 2022-11-11:
This is indeed a strange edge case. Looks like we got at least 
- 2 successful installations on Raspberry PIs that worked without adding www-data to lndg group and 
- 2 installations (ohohora and myself) on Ubuntu 22.04 that only worked by adding www-data to lndg group

I'm just guessing but it could be OS related.

VajraOfIndra, 2022-11-11:
I'll make www-data a member of the lndg group in the guide, like that it works for any platform.

I was thinking of the security consideration of giving www-data some lndg group permissions.. but I don't think it creates any real secuirty concerns.

closed 2022-11-11 by VajraOfIndra

ohohora, 2022-11-11:
OK. maybe related to OS like blckbx said....? Anyway, I hope official guide of LNDg will come out well~!!!

ohohora, 2022-11-11:
OK. Maybe related to OS like blckbx said...? Anyway, I wish official LNDg guide will come out well~!!!

PR #1070 (2022-11-11): New bonus guide: LNDg
#### What

LNDg is a lite GUI web interface to help you manually manage your node and automate operations such as rebalancing, fee adjustemnts and channel node opening.

Has been requested by the community: e.g. https://t.me/raspibolt/12058

#### How

A new bonus guide in the +Lightning section

#### Scope

- [ ] significant change to core configuration
- [x] independent bonus guide
- [ ] simple bug fix

Fixes #1125

#### Test & maintenance

How can the changes be tested? Is there ongoing maintenance effort? If this is a bonus guide: are you willing to update it from time [... cut]