raspibolt/raspibolt #1127: no liquidity management with main guide & RTL?

uskrusader, 2022-10-18:
Greetings, I followed the guide, and kept everything up to date since starting this project almost a year ago.  I haven't made much progress with my lightning node.  I have a few channels, but I only have inbound liquidity from a friend's node.  I can't seem to do "loop out" or do "circular rebalance" with other channels from within RTL or Zeus.  

Is lightning terminal the way to go?  Will this allow "Loop out" on RTL?  
Does this work in conjunction with RTL?  Or does it mostly replace RTL?

tanks!

VajraOfIndra, 2022-10-18:
> Is lightning terminal the way to go? Will this allow "Loop out" on RTL?
Lightning Terminal would allow you to do Loop out yes, either with the Lightning Terminal CLI or their web UI or directly from RTL if you have the Loop daemon running and configure RTL. [The guide](https://raspibolt.org/guide/bonus/lightning/lightning-terminal.html) has not been looked at for a while but it should still explains how to install and use all of the above.

> Does this work in conjunction with RTL? Or does it mostly replace RTL?
Loop can be done from within RTL but it does not replace RTL as RTL has many features besides looping out.

Do you want your node to be tuned to be used:
1) Simply to make and receive payments? or;
2) To be used for routing?

If 1), then looping out could be a good option. It's easy and quick. 
But looping is expensive (Lightning Labs take a fee and the routing node routing your payment to LOOP charge very hefty fees of ~2000 ppm).  

So if you aim at 2) (being a routing node), or if you want to save on sats, better avoid looping out if you can.

In that case, the best way to manage liquidity by circular rebalancing your channels using tools like:
- [Balance of Satochis](https://raspibolt.org/guide/bonus/lightning/balance-of-satoshis.html): manual rebalances, CLI-only
- [rebalance-lnd](https://raspibolt.org/guide/bonus/lightning/rebalance-lnd.html): manual or semi-automated rebalances, CLI-only
- [LNDg](https://github.com/raspibolt/raspibolt/blob/a3cc0680535672e13f50f7309f4166d326de56c4/guide/bonus/lightning/lndg.md): manual or fully-automated rebalances, web UI-only (but the guide is still a draft, not to be used for now)

Before using the automatd features of rebalance-lnd and LNDg it's good to do some manual rebalances to understand what's going on and avoid paying too many rebalancing fees. For this, I think Balance of Satoshis is a good tool to start with.

Moreover, Balance of Satoshis allow you to:
- Open dual-funded channels: to get some inbound liquidity from the get-go, and more capacity on the node
- Batch open channels: cheaper than making a tx for each individual channel

Just a few thoughts, happy to help if you get stuck somewhere.

uskrusader, 2022-10-18:
fantastic!, thanks for the great info.  It sounds like Balance of Satoshis is the way.  I'm in for the long run, payments is primary, but volume will increase.  Routing is secondary for now, but still an objective.  thanks again!

closed 2022-10-18 by uskrusader