raspibolt/raspibolt #1132: [FEATURE REQ] Create a bonus guide for alternative hardware/OSes

VajraOfIndra, 2022-10-21:
### What & why

As Raspberry Pis have become [expensive](https://www.cnx-software.com/2022/02/24/raspberry-pi-4-price-expensive-europe/), more and more users are looking into cheap alternative platforms to run their Bitcoin and Lightning nodes.

Also, some users might want better hardware to run a more performant routing node.

Although, the guide is mostly compatible with any Debian-based OSes, a few things have to be considered when setting up and running a node on an alternative hardware and OS.

A [poll ](https://t.me/raspibolt/12643) on the TG group shows that such a guide would be welcomed by our community:
[image]

### How

The bonus guide could have the following goals:
1. A list of suggested alternative hardware (e.g. [refurbished ThinkPad](https://twitter.com/i/communities/1563029300911058944))
2. A list of suggested OSes (e.g. Ubuntu Server, etc)
3. A list of guidelines on how to adapt the RaspiBolt guides to alternative hardware (e.g. check CPU architecture, select appropriate binary release etc)
4. Some tips on how to select the appropiate specs (CPU, RAM, SSD type, RAID) for one's own needs (budget? is it for a routing node? small or large node? is power consumption important? etc)

This [Raspiblitz guide ](https://github.com/rootzoll/raspiblitz/tree/dev/alternative.platforms) is a great resource we could use for inspiration.

[image]

VajraOfIndra, 2022-10-22:
To add: 
- Need to create the "admin" user on some OSes (e.g. Ubuntu, see [issue ](https://t.me/raspibolt/14280) here)

Fiach-Dubh, 2022-11-01:
I'd like to cosign this idea as I had a similar thought. installing debian on a old laptop or desktop computer should be able to follow the same instructions.

But it's not clear in the guide that using other hardware is a viable option for end users. It would help if this were added as a note.

twofaktor, 2022-12-06:
Hi! A have a personal adaptation called MiniBolt in a separate repo, check out this: 

https://github.com/twofaktor/minibolt
https://twofaktor.github.io/minibolt

We could collaborate together with different new guides/features, in addition, to integrating both guides as bonus guides 🤝

closed 2023-03-15 by github-actions[bot]