raspibolt/raspibolt #1138: [FEATURE REQ] Add bonus guide on how to run Tor relay

kristapsk, 2022-10-29:
### What & why

Tor network is under DDoS attack for some time. More people running Tor relays would help. People having good enough Internet connection and powerful enough hardware could help Tor network (and also Bitcoin, as a lot of privacy stuff there currenty depends on Tor) by doing that.

VajraOfIndra, 2022-11-04:
What are the risks associated with it?

@kristapsk: wdyt of #1110 about setting up a Tor bridge? I can't review it as I don't know enough about Tor and networking.

twofaktor, 2023-01-03:
I'm running a obfs4 Tor bridge: https://metrics.torproject.org/rs.html#details/C390F4996FE42E2623B47FB2FA96B2D9A771F358

Following official docs, they say no risk associated with running a Tor bridge because summarizing Tor bridges are nodes in the network that are not listed in the public Tor directory --> ref: https://community.torproject.org/relay/types-of-relays/

blckbx, 2023-01-04:
Something to keep in mind: Running a relay requires some serious preconditions that may cause trouble for a lot of home setups. Here's one of them: `A non-exit relay should be able to handle at least 7000 concurrent connections.` ([src](https://community.torproject.org/relay/relays-requirements/))

twofaktor, 2023-01-06:
In my case, my Tor bridge: https://metrics.torproject.org/rs.html#details/C390F4996FE42E2623B47FB2FA96B2D9A771F358
Last log:
`Jan 06 18:17:36 Tor[17986]: Heartbeat: Tor's uptime is 18 days 0:00 hours, with 380 circuits open. I've sent 266.22 GB and received 266.67 GB. I've received 52843 connections on IPv4 and 0 on IPv6. I've made 124867 connections with IPv4 and 0 with IPv6.`
No problem was detected with my router or local internet connection in my house or any device.

I'm using Technicolor TC7230

closed 2023-03-09 by kristapsk