raspibolt/raspibolt #1139: running lightning cli via admin not working "lightning-cli: Connecting to 'lightning-rpc': Connection refused"

ericsupreme, 2022-10-29:
so i followed the instructions per: 
https://raspibolt.org/guide/bonus/lightning/cln.html#allow-user-admin-to-work-with-cln

and i'm still receiving "lightning-cli: Connecting to 'lightning-rpc': Connection refused"

i restarted services ( bitcoind and lightningd) i reboot nothing...any suggestions?

thanks in advanced

zynos, 2022-11-03:
Can you share your config file? (https://raspibolt.org/guide/bonus/lightning/cln.html#config-file)
Also which outputs do you get when you run
`sudo journalctl -f -u lightningd`
and
`lightning-cli getinfo`
?

blckbx, 2022-11-09:
Most likely bitcoind has not finished starting up or chain is not completely synced.

closed 2023-01-15 by github-actions[bot]

Ripley81, 2023-02-07:
This is typically because Bitcoind or LND isn't running.

Verify that bitcoin is running. As admin user:
        bitcoin-cli getblockchaininfo

if this returns an RPC error, then it's not running and you need to restart it.
        sudo systemctl restart bitcoind

then enable, start and unlock LND
	sudo systemctl enable lnd
	sudo systemctl start lnd
	systemctl status lnd

remember to also restart Electrum server and RTL and Blockexplorer.