raspibolt/raspibolt #1141: [BUG]

KodonFrost, 2022-10-30:
### Describe the bug
github has replaced "master" with "main" for the branch in new repos.
This breaks the guide for the channel backup.

### Location in guide
[https://raspibolt.org/guide/lightning/channel-backup.html](https://raspibolt.org/guide/lightning/channel-backup.html)

[https://raspibolt.org/guide/lightning/channel-backup.html#create-script](https://raspibolt.org/guide/lightning/channel-backup.html#create-script)
1: In the scb-backup script file replace "git push --set-upstream origin master" with "git push --set-upstream origin main"

[https://raspibolt.org/guide/lightning/channel-backup.html#test](https://raspibolt.org/guide/lightning/channel-backup.html#test)
2: For the push of the test dummy file replace "git push --set-upstream origin master" with "git push --set-upstream origin main"

### Expected behavior
Push fails with "master" as it is no longer used.
Using "main" instead will restore functionality.

### Environment
All.

### Additional context

One additional option to think about:
In the scb-backup script I added the force flag to the "git push **-f** --set-upstream origin main" command.
This makes sure that the new backup file will be pushed, even if something else made changes in the meantime.
Forcing the push avoids problems with the upload that might otherwise go unnoticed until you need the backup and then realize you don't have one.

The problem I had without the -f flag:
```
To github.com:xxx/remote-lnd-backup.git
 ! [rejected]        main -> main (fetch first)
error: failed to push some refs to 'github.com:xxx/remote-lnd-backup.git'
hint: Updates were rejected because the remote contains work that you do
hint: not have locally. This is usually caused by another repository pushing
hint: to the same ref. You may want to first integrate the remote changes
hint: (e.g., 'git pull ...') before pushing again.
hint: See the 'Note about fast-forwards' in 'git push --help' for details.
```

kristapsk, 2022-10-30:
Cannot reproduce with newly created test repo. Default branch name is just a recommendation, you can actually use whatever name you want.
```
$ git clone git@github.com:kristapsk/testrepo.git
Cloning into 'testrepo'...
warning: You appear to have cloned an empty repository.
$ cd testrepo
$ git status
On branch master

No commits yet

nothing to commit (create/copy files and use "git add" to track)
$ touch kaka
$ git add kaka
$ git commit -m "test"
[master (root-commit) d3436b3] test
 1 file changed, 0 insertions(+), 0 deletions(-)
 create mode 100644 kaka
$ git push -u origin master
Enumerating objects: 3, done.
Counting objects: 100% (3/3), done.
Writing objects: 100% (3/3), 1.34 KiB | 1.34 MiB/s, done.
Total 3 (delta 0), reused 0 (delta 0), pack-reused 0
To github.com:kristapsk/testrepo.git
 * [new branch]      master -> master
Branch 'master' set up to track remote branch 'master' from 'origin'.
```

KodonFrost, 2022-10-30:
Yes, you can name the branch whatever you like.

After creating a new account github account the new default branch name is `main` and not `master`.
By following the guide to the letter, you'll receive `error: src refspec master does not match any` when attempting `git push --set-upstream origin master`.

The guide works with the default values when creating the repo. The new default branch name is `main`.

If `master` is the desired branch name, a new user can not simply follow the guide to create a new repo, but has to redefine the default branch name first.
Otherwise the script will fail to run and it's not really obvious to find the error. Especially if the user has no experience with github.

[image]

kristapsk, 2022-11-24:
> After creating a new github account the new default branch name is `main` and not `master`.

Forgot that I have especially changed my github settings so that for new repos it defaults to `master`, not `main`, like it was before SJW attack. 
[image]

kristapsk, 2023-01-23:
Should not be closed, it's an issue for most of the people.

kristapsk, 2023-03-24:
I think it's still an issue and should not be closed.

kristapsk, 2023-07-29:
This is still an issue.

closed 2023-10-11 by github-actions[bot]