raspibolt/raspibolt #1143: Use of various users

AngeloMetal, 2022-10-31:
I've been struggle with logging in, logging out, granting these rights, etc., to users `admin`, `bitcoin`, `electrs` etc., but at this point I don't understand why I'm doing these things. And, what I like in DIY stuff, is that I can understand every step in a satisfactory manner. 

My question is: What would be the problem if I made the raspibolt installation in just one user (`admin`)? In `Bitcoin client`, you describe that this happens for security reasons.

> Create the bitcoin user
>
> The Bitcoin Core application will run in the background as a daemon and use the separate user “bitcoin” for security reasons. This user does not have admin rights and cannot change the system configuration

What's the problem with granting `bitcoin` the rights to change the system configuration? If an attacker gains access to admin, can't he access the rest users as well? Is it less probable to happen with `admin`?

VajraOfIndra, 2022-11-01:
If an attacker get access to the "admin" user, then yes, they could do anything they want.

The point of running each program with their own users is that if that program contains malicious code, the malicious code will only be able to do, at most, what the owner of that program can do.

 For example, if you update Electrs with some malicious code in it, the hacker might be able to:
- "read, write, execute" anything related to Electrs
- but only "read" files in the Bitcoin folder
- and wouldn't be able to do much at all with the folders of any other users (e.g. LND)

At least, that's my understanding, someone else might be able to explain this more accurately or in a better way.

AngeloMetal, 2022-11-02:
Yes, but there's minimum risk in allowing tested, peer reviewed software such as Bitcoin Core, Core Lightning, Electrs, or JoinMarket to access the admin, while there are some high chances you screw it up with users provided that you're not an experienced linux user.

kristapsk, 2022-11-02:
Using different UNIX users for different software is just a common sense and good security practice. It does not mean it will always protect you, as privilege escalation bugs happen through which you increase your given permissions in a system, but still gives you some protection. Otherwise you wouldn't even need `admin` user and could run everything as a `root`.

> If an attacker get access to the "admin" user, then yes, they could do anything they want.

Not really, as `sudo` will ask for password. If attacker gains access to some user via some vulnerability doesn't mean he knows the password.

AngeloMetal, 2022-11-02:
> Not really, as sudo will ask for password. If attacker gains access to some user via some vulnerability doesn't mean he knows the password.

Which means that an ``admin`` with strong password is enough?

kristapsk, 2022-11-02:
> Which means that an `admin` with strong password is enough?

No, if you run everything under the same user, then attacker gaining access to the system through one program will gain access to data, configuration, memory contents, etc of all the other programs, without a need to look for some other vulnerabilitties in the system.

Remember that we are talking here about potentially stealing money not just some private nude pictures.

closed 2022-11-24 by kristapsk