raspibolt/raspibolt #1165: [FEATURE REQ] Add the new headless server "Sparrow Terminal" installation instructions to Sparrow guide

VajraOfIndra, 2022-11-11:
Sparrow can now be run on headless server (like a Raspberry Pi). The new feature is called [Sparrow Terminal](https://twitter.com/SparrowWallet/status/1585597388751790084).

Probably useful enought to add to the main Sparrow guide. Otherwise as a bonus guide.

DanielPX1, 2022-11-18:
working on it 👍

VajraOfIndra, 2022-11-18:
> working on it 👍

In your PR, you can add a keyword + the Issue number to close this issue automatically when the corresponding PR is merged. It's a nice feature of GitHub :)

E,g. 

Fixes #1165

The list of keywors is:

    close
    closes
    closed
    fix
    fixes
    fixed
    resolve
    resolves
    resolved

So for closing a feature request, maybe: Resolves #1165 or Closes #1165 would fit well,. (more than Fixes #1165 which would be more suitable to a bug fix.

More info here: https://docs.github.com/en/issues/tracking-your-work-with-issues/linking-a-pull-request-to-an-issue

satsHIVE, 2022-12-02:
This should be resolved by https://github.com/raspibolt/raspibolt/pull/1169. No?

twofaktor, 2022-12-02:
> This should be resolved by #1169. No?

I agree, closed as completed!

twofaktor, 2022-12-02:
I agree, closed as completed!

closed 2022-12-02 by twofaktor