raspibolt/raspibolt #1206: [BUG] : Port conflict between Homer and plugin for CLN: RTL/c-lightning-Rest

vicariousdrama, 2022-12-06:
### Describe the bug

In the guides, both of these services are setup to use port 4091.  This will cause either nginx or the node process for c-lightning rest plugin documentation to fail depending on which is started first

### Location in guide

For Homer, this is established in these places
* UFW setting
  * https://github.com/raspibolt/raspibolt/blame/master/guide/bonus/raspberry-pi/homer.md#L61
* Guidance for NGINX
  * https://github.com/raspibolt/raspibolt/blame/master/guide/bonus/raspberry-pi/homer.md#L61
* listen directive in server block of homer-ssl.conf
  * https://github.com/raspibolt/raspibolt/blame/master/guide/bonus/raspberry-pi/homer.md#L150-151
* Guidance for initial testing access in browser
  * https://github.com/raspibolt/raspibolt/blame/master/guide/bonus/raspberry-pi/homer.md#L443
* Guidance for testing access in browser after started as a service
  * https://github.com/raspibolt/raspibolt/blame/master/guide/bonus/raspberry-pi/homer.md#L496
* Listing firewall rules for Homer
  * https://github.com/raspibolt/raspibolt/blame/master/guide/bonus/raspberry-pi/homer.md#L660

For Core Lightning with the rest plugin for Ride the Lightning
* DOCPORT assignment in lightningd config file
  * https://github.com/raspibolt/raspibolt/blame/master/guide/bonus/lightning/cln.md#L433
* And the rest config file
  * https://github.com/raspibolt/raspibolt/blame/master/guide/bonus/lightning/cln.md#L455
* Sample log output
  * https://github.com/raspibolt/raspibolt/blame/master/guide/bonus/lightning/cln.md#L489

### Expected behavior

There should be no port conflicts for any of the guide combinations used.  
Each guide should have its own unique port.  
Recommend creation of a PORTS.MD file or other central point to quickly reference for what each of the service ports are.

### Additional context

To mitigate this, the port can be changed in either Homer or Core Lightning rest plugin.  We resolved in homer-ssl.conf by using a different port.  Port 4873 was chosen (based on 4000 + BIP39 word for "home" = 873).

kristapsk, 2022-12-06:
> Recommend creation of a PORTS.MD file or other central point to quickly reference for what each of the service ports are.

Agree!

kristapsk, 2023-02-04:
Should not be closed.

kristapsk, 2023-04-03:
> Enabled Wiki setting in repository and started an initial page for used ports: https://github.com/raspibolt/raspibolt/wiki/Ports

I think it would be better to keep this as a doc in git repo instead. Wiki is GitHub specific thing, removes flexibility to potentially move from GitHub to some other platform in future.

dutu, 2023-06-05:
A fix is proposed here: https://github.com/raspibolt/raspibolt/pull/1260, but this PR had no review or comment in two months.

closed 2023-08-11 by github-actions[bot]