raspibolt/raspibolt #1217: [UPDATE] Node.js installation instructions

purpleKarrot, 2022-12-29:
On the pages "Bitcoin/Blockchain explorer" and "Lightning/Web app", the installation instructions for Node.js claim that **16** is the latest stable version, which is not up to date.

Instead of changing to the *actual* latest stable version (which is 19 at the time of this writing), you might also use the the value `current` (or `lts`, for long term support):

```
curl -fsSL https://deb.nodesource.com/setup_current.x | bash -
```

https://github.com/nodesource/distributions/blob/master/README.md#debinstall

twofaktor, 2022-12-30:
Hello, thanks for your contribution, following the official webpage, the latest nodejs version is 18.12.1 LTS

https://nodejs.org/en/

kristapsk, 2022-12-30:
Blindly upgrading everything to the latest version might not be what you always want. Not so familiar with Node.js, but aren't there possible incompatibilities between major releases that can break some older software?

purpleKarrot, 2022-12-30:
> Hello, thanks for your contribution, following the official webpage, the latest nodejs version is 18.12.1 LTS
> 
> 
> 
> https://nodejs.org/en/

Look closer. The latest version is 19.3.0, version 18.12.1 is maintained as LTS.

Nodesource provides installation scripts for both versions.

blckbx, 2023-01-05:
> Blindly upgrading everything to the latest version might not be what you always want. Not so familiar with Node.js, but aren't there possible incompatibilities between major releases that can break some older software?

I'm running Thunderhub, RTL, BoS with Node.JS LTS without problems. The only thing I've seen so far was Thunderhub not starting on Node.JS v19. So I would consider using LTS as safe.

felipeucc, 2023-01-13:
Samourai requires version 14 or 16. I was doing all my installation with version 19 and during the installation of Samourai I got an error message that the node version was not compatible

closed 2023-03-21 by github-actions[bot]