raspibolt/raspibolt #1230: [BUG] 404 error on Signature check of Bitcoin Core

Ripley81, 2023-01-14:
### Description
I'm installing a Bitcoin node as per the Raspibolt Guide and getting a 404 error at the step of verifying signatures of the Bitcoin Core download, and cannot see the corresponding folder structure on Github/Bitcoin.

admin@raspibolt:~ $ wget https://raw.githubusercontent.com/bitcoin/bitcoin/master/contrib/builder-keys/keys.txt
--2023-01-14 19:44:44--  https://raw.githubusercontent.com/bitcoin/bitcoin/master/contrib/builder-keys/keys.txt
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.111.133, 185.199.108.133, 185.199.109.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.111.133|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2023-01-14 19:44:45 ERROR 404: Not Found.

### Location in guide
https://raspibolt.org/guide/bitcoin/bitcoin-client.html#signature-check

### Environment
 - Hardware platform: Raspberry Pi 4b
 - Operating system: Raspberry Pi OS 64bit
 - Version: Debian GNU/Linux 11 (bullseye)

kristapsk, 2023-01-14:
Builder keys file were removed with https://github.com/bitcoin/bitcoin/pull/26598.

Some keys can be fetched from https://github.com/bitcoin-core/guix.sigs/tree/main/builder-keys instead.

kristapsk, 2023-01-14:
There's related commit for Minibolt, guess we can just use the same - https://github.com/twofaktor/minibolt/commit/4d03e98702c2d11405f37484adc5658cac7bf525. Ping @twofaktor !

twofaktor, 2023-01-14:
Is already resolved here, give me few days to suggest a PR for Raspibolt
 https://twofaktor.github.io/minibolt/guide/bitcoin/bitcoin-client.html#signature-check

Ripley81, 2023-01-14:
Appreciated

s373nZ, 2023-01-27:
`jq` needs to be listed as an install dependency. I don't have a better solution, but the command is pretty gnarly...

twofaktor, 2023-01-28:
> `jq` needs to be listed as an install dependency. I don't have a better solution, but the command is pretty gnarly...

`jq` is already listed as an install dependency:
https://minibolt.info/guide/system/configuration.html#system-update

If you have a better solution let us know

s373nZ, 2023-01-28:
@twofaktor Oh, I wound up here from the Raspibolt guide, following instructions to upgrade bitcoind to `14.0.1`.

https://github.com/raspibolt/raspibolt/blob/master/guide/bitcoin/bitcoin-client.md?plain=1#L534

In terms of a cleaner solution, are there restrictions to maintining a `/scripts` directory in the Raspibolt repo for helper functionality, or is that something we'd rather be staying away from?

s373nZ, 2023-01-28:
Initial branch: https://github.com/raspibolt/raspibolt/compare/master...s373nZ:raspibolt:bitcoind-key-import-fix

twofaktor, 2023-01-30:
New PR opened https://github.com/raspibolt/raspibolt/pull/1238

PR #1238 (2023-02-01): [UPDATE MAIN GUIDE] Update with fixing signature check Bitcoin Core + dependencies update
#### What

Adaptation to the new Signature verification process for Bitcoin Core 🔑 – Based the on Guix attestation method to release new Bitcoin Core releases with Bitcoin Core developer's signatures

GitHub repo of Bitcoin Core release attestations (Guix): https://github.com/bitcoin-core/guix.sigs

History:
PR that caused the broke and obsolescence of the old signature verification process: https://github.com/bitcoin/bitcoin/pull/26598

New GitHub folder of the Bitcoin Core repo that stores the signature [... cut]