raspibolt/raspibolt #1232: After updating lnd getting 'Exec Format Error"

Quenos, 2023-01-22:
### Describe the issue
Followed all the steps on: https://raspibolt.org/guide/lightning/lightning-client.html#installation (except timestamp check because ots is not installed)
Running lnd --version 
> -bash: /usr/local/bin/lnd: cannot execute binary file: Exec format error

### Location in guide
https://raspibolt.org/guide/lightning/lightning-client.html#installation

### Screenshots
N/A

### Environment
 - Raspberry Pi 4
 - Raspbian GNU/Linux 11 (bullseye)
 - Version: $ lsb_release -d
Traceback (most recent call last):
  File "/usr/bin/lsb_release", line 25, in <module>
    import lsb_release
ModuleNotFoundError: No module named 'lsb_release'

### Additional context
N/A

Quenos, 2023-01-22:
The problem lays in the fact that the ARM64 is not working. Taking the V6 version works wonderfully.

closed 2023-01-22 by Quenos

kristapsk, 2023-01-22:
You should run 64-bit OS, not 32-bit OS.