raspibolt/raspibolt #1234: Torifying PuTTY - connection error

captainbluebaer, 2023-01-29:
i was installing the SSH remote access through Tor (optional).
described here:
https://raspibolt.org/guide/raspberry-pi/privacy.html

SSH Server:
i got the TOR address and it looks like it's working until here.
now i am stuck in the next section:

SSH client:
The pictures of the guide on the link below are broken. So i'm stuck here by putting the right configuration in PuTTY.
https://gitlab.torproject.org/legacy/trac/-/wikis/doc/TorifyHOWTO/Putty

i get now the following error when i want to connect to the server:
PuTTY fatal error: network error: connection refused.

a connection without tor is possible. i think i only have to put the right configuration in Putty to connect via Tor.

i hope somebody could help me with this issue.

thank you !!

fritzjohnbachman, 2023-03-04:
You are not alone in needing help on this issue.

fritzjohnbachman, 2023-03-04:
Proxy Settings look like this:
[image]

... it's the Session settings messing me up

fritzjohnbachman, 2023-03-04:
Finally figured this out.
Need to be running the Tor browser in Windows (minimized or whatever).
The command "sudo cat /var/lib/tor/hidden_service_sshd/hostname" on your RaspiBolt gave you your Tor connection address
In PuTTY, put only this resulting .onion address in the Host Name field (indicated by "$<Tor_hostname>.onion")
[image]
Leave Port as 22
Set Proxy settings as previously mentioned. SOCKS 5, localhost, 9150, DNS Yes
Click Open to open connection
Proxy SOCKS 5 connection will prompt for username and password. Use 'admin' and password [A] from instructions (Master User Password.)
SSH connection will be made to your .onion address and prompt you again for username and password. Again, use 'admin' and password [A] from instructions.
Console interaction will end up looking something like this:
[image]

I hope this works for you as well.

captainbluebaer, 2023-03-20:
Hi fritzjohnbachman. Great!! Your walkthrough worked for me! thank you for your help!!