raspibolt/raspibolt #1235: JAM - No connection to backend: Gateway Timeout - ECONNREFUSED

SpartanBit, 2023-01-29:
Hi all, I have followed the JAM guide described here:
https://raspibolt.org/guide/bonus/bitcoin/Jam.html

however the webui on the local network shows the following message: No connection to backend: Gateway Timeout. 
This seems a very similar issue: https://github.com/joinmarket-webui/jam/issues/88

If I have a look at the logs, every time I try to reach JAM I get the following:

Error occurred while proxying request 192.168.1.18:4020 to undefined [ECONNREFUSED] (https://nodejs.org/api/errors.html#errors_common_system_errors)
Error occurred while proxying request 192.168.1.18:4020/api/v1/session to https://localhost:28183/ [ECONNREFUSED] (https://nodejs.org/api/errors.html#errors_common_system_errors)
Error occurred while proxying request 192.168.1.18:4020/api/v1/wallet/all to https://localhost:28183/ [ECONNREFUSED] (https://nodejs.org/api/errors.html#errors_common_system_errors)

Not sure what's causing it, but i tried to get to the root cause of this, and the only thing i've found is that some of the other guides do specify in the .env file not only the port but also the host/server at 127.0.0.1:... 
see https://raspibolt.org/guide/bonus/lightning/thunderhub.html#configuration or https://raspibolt.org/guide/bitcoin/blockchain-explorer.html#configuration

Could that be the cause behind this problem, and if so, how could i resolve it?

I would be truly grateful if someone could help.
Thanks!

theborakompanioni, 2023-03-13:
Please see https://github.com/joinmarket-webui/jam/pull/611#issuecomment-1457995514
Problem might be that Jam proxies to "localhost", which resolves to ipv6 `::1`, but nginx only listens on ipv4.
Would changing the guide https://github.com/raspibolt/raspibolt/blob/master/guide/bonus/bitcoin/Jam.md?plain=1#L128
```patch
-  listen 4020 ssl;
+  listen [::]:4020 ssl;
```
solve the issue?

dutu, 2023-04-25:
> Please see [joinmarket-webui/jam#611 (comment)](https://github.com/joinmarket-webui/jam/pull/611#issuecomment-1457995514) Problem might be that Jam proxies to "localhost", which resolves to ipv6 `::1`, but nginx only listens on ipv4. Would changing the guide https://github.com/raspibolt/raspibolt/blob/master/guide/bonus/bitcoin/Jam.md?plain=1#L128
> 
> ```diff
> -  listen 4020 ssl;
> +  listen [::]:4020 ssl;
> ```
> 
> solve the issue?

I have tested this and it does not solve the issue. 

In fact it cannot solve it. The problem is not on port 4020, but the issue is with application accessing https://localhost:28183, https://localhost:28283 and http://localhost:62601

dutu, 2023-04-25:
See https://github.com/joinmarket-webui/jam/issues/608#issuecomment-1522515632 for a solution

closed 2023-07-02 by github-actions[bot]