raspibolt/raspibolt #1237: "unable to find zmqpubrawblock in config, cannot start w/o RPC connection"

smartm0use, 2023-01-29:
I'm following this [guide](https://raspibolt.org/guide/lightning/lightning-client.html) to install **lnd** on my machine with Ubuntu 20.04.

When I try to run `lnd` I get this error:

**failed to load config: ValidateConfig: unable to load RPC credentials for bitcoind: unable to extract RPC credentials: unable to find zmqpubrawblock in config, cannot start w/o RPC connection**

I made some searches and I see that I have to add the following lines to the lnd configuration file:

```
[Bitcoind]
bitcoind.zmqpubrawblock=127.0.0.1:28332
bitcoind.zmqpubrawtx=127.0.0.1:28333
```

But when I try to run again `lnd` I get this:

**failed to load config: ValidateConfig: unable to load RPC credentials for bitcoind: please set bitcoind.rpcuser and bitcoind.rpcpass (or bitcoind.rpccookie) together with bitcoind.zmqpubrawblock, bitcoind.zmqpubrawtx**

I also tried to add these lines to bitcoin.conf file of my Bitcoin Core:

```
zmqpubrawblock=127.0.0.1:28332
zmqpubrawtx=127.0.0.1:28333
```

Have I to set an user and password for rpcuser and rpcpass? Why? Wasn't possible to get access credentials thanks to .cookie file in the Bitcoin data directory?
I thought that the way lnd could connect to Bitcoin Core daemon was the same or something similar to how Electrum Server does, I mean just look for bitcoind instance and work. Am I wrong?

twofaktor, 2023-01-30:
You didn't follow the guide, there isn't any configuration on LND same this:
 
> [Bitcoind]
> bitcoind.zmqpubrawblock=127.0.0.1:28332
> bitcoind.zmqpubrawtx=127.0.0.1:28333

https://raspibolt.org/guide/lightning/lightning-client.html#configuration

You need to follow this section:

https://raspibolt.org/guide/bitcoin/bitcoin-client.html#configuration

And restart bitcoind

Please follow the RaspiBolt guide

smartm0use, 2023-01-30:
> You didn't follow the guide, there isn't any configuration on LND same this:
> 
> > [Bitcoind]
> > bitcoind.zmqpubrawblock=127.0.0.1:28332
> > bitcoind.zmqpubrawtx=127.0.0.1:28333
> 
> https://raspibolt.org/guide/lightning/lightning-client.html#configuration
> 

I know there is indication on these lines in the guide, I found them after I made some searches (surfing Google), as I said.

> You need to follow this section:
> 
> https://raspibolt.org/guide/bitcoin/bitcoin-client.html#configuration
> 
> And restart bitcoind
> 
> Please follow the RaspiBolt guide

I did, but I'm getting a new error:

**failed to load config: ValidateConfig: unable to load RPC credentials for bitcoind: unable to extract RPC credentials: unable to find rpcuser in config, cannot start w/o RPC connection**

I think I have to give up, because I'm not on Raspbian OS, as you recommended [here](https://github.com/raspibolt/raspibolt/issues/1236).

twofaktor, 2023-01-30:
> 

Did you skip this step? https://raspibolt.org/guide/raspberry-pi/operating-system.html

Start again, follow all steps, (since the first step) and understand each of them

closed 2023-04-07 by github-actions[bot]