raspibolt/raspibolt #1239: [BUG / BUMP] JoinMarket bump version to 0.9.9

sig-noi, 2023-01-31:
### Describe the bug
Guide needs to be updated for version 0.9.9
The process of the upgrade, for me at least, had me wanting better step-by-step instructions. Issues included:
- “Install JoinMarket” link at bottom of current joinmarket.md guide is broken 
- ots verification didn’t work from the joinmarket user, had to exit to admin and then go back to joinmarket. Not sure why.
- need to add a step to rename the old joinmarket.cfg file before running wallet-tool.py to create new joinmarket.cfg file

### Location in guide
(https://raspibolt.org/guide/bonus/bitcoin/joinmarket.html#introduction)

### Expected behavior
Expected installation to be a little tricky, would like to make it less tricky.

### Screenshots
n/a

### Environment
Rasberry Pi 4, RaspiBolt v3, fully updated

Mynima, 2023-01-31:
Have added a PR for the with a few of the above items (if you've get chance to review it that would be great). My testing found that the ots verification seemed to work fine from the joinmarket user, did you make sure to perform the following step to create the symbolic link `ln -s /data/bitcoin /home/joinmarket/.bitcoin` when you went through the process first time?

Could you elaborate regarding the rename/backup of the old joinmarket.cfg file. Was you intention an additional couple of steps in the upgrade section?

sig-noi, 2023-02-02:
Thanks for the tip about that symbolic link. Now the ots verification works for me. I commented on your PR to suggest what I meant about the joinmarket.cfg file. Haven’t fully tested the PR since I’d already upgraded, but like the VERSION variable and generally ACK.

Mynima, 2023-02-02:
> Thanks for the tip about that symbolic link. Now the ots verification works for me. I commented on your PR to suggest what I meant about the joinmarket.cfg file. Haven’t fully tested the PR since I’d already upgraded, but like the VERSION variable and generally ACK.

excellent, thanks for the feedback.

kristapsk, 2023-02-03:
Resolved by #1240.

closed 2023-02-03 by kristapsk