raspibolt/raspibolt #1241: New release of c-lightning-REST v0.10.0

kristapsk, 2023-02-09:
There is new release of c-lightning-REST v0.10.0 - https://github.com/Ride-The-Lightning/c-lightning-REST/releases/tag/v0.10.0.

Core Lightning bonus guide likely needs to be updated. By somebody who uses CLN with RaspiBolt.

kristapsk, 2023-02-11:
Ping @blckbx

kristapsk, 2023-02-22:
There's some breaking changes in Core Lightning v22.11.0+, c-lightning-REST needs to be updated to v0.10.1. https://github.com/Ride-The-Lightning/c-lightning-REST/releases/tag/v0.10.1

blckbx, 2023-02-22:
PR updated, hotfix relates to offer api.

kristapsk, 2023-02-22:
> PR updated, hotfix relates to offer api.

Noticed Core Lightning bonus guide has v0.11.2 currently. Guess that also should be updated to newer version.

blckbx, 2023-02-22:
> Noticed Core Lightning bonus guide has v0.11.2 currently. Guess that also should be updated to newer version.

Yes, that's mentioned in CLN-rest PR. PR is still waiting for approval (#1210)

kristapsk, 2023-02-22:
> > Noticed Core Lightning bonus guide has v0.11.2 currently. Guess that also should be updated to newer version.
> 
> Yes, that's mentioned in CLN-rest PR. PR is still waiting for approval (#1210)

Sorry, missed that!

blckbx, 2023-02-22:
Nobody's fault. I was hoping for some testers.

PR #1246 (2023-03-27): [bonus guide update] c-lightning-rest v0.10.1
### What

Update of c-lightning-rest interface to version 0.10.1. 

### How

Local test of c-lightning-rest v0.10.1 installation guide w/ RTL v0.13.3.

### Scope

- [ ] significant change to core configuration
- [x] independent bonus guide update
- [ ] simple bug fix

### Test

Locally tested on amd64 bolt setup w/ CLN 22.11.1 and RTL v0.13.3

### Dependency

CLN 22.11.1 should ideally be merged first ( PR #1210 )

Fixes #1241
closed 2023-03-27 by blckbx