raspibolt/raspibolt #1254: [BUG]

ywiyogo, 2023-04-02:
### Describe the bug
I did the guide until https://raspibolt.org/guide/raspberry-pi/privacy.html#ssh-remote-access-through-tor-optional.
Everthing works good, until I try to connect from another Ubuntu22 computer with `torify ssh admin@....`. It returns:

```
1680398012 WARNING torsocks[13080]: [syscall] Unsupported syscall number 39. Denying the call (in tsocks_syscall() at syscall.c:604)
1680398012 WARNING torsocks[13080]: [syscall] Unsupported syscall number 39. Denying the call (in tsocks_syscall() at syscall.c:604)
1680398012 WARNING torsocks[13080]: [syscall] Unsupported syscall number 39. Denying the call (in tsocks_syscall() at syscall.c:604)
1680398012 WARNING torsocks[13080]: [syscall] Unsupported syscall number 39. Denying the call (in tsocks_syscall() at syscall.c:604)
1680398012 WARNING torsocks[13080]: [syscall] Unsupported syscall number 39. Denying the call (in tsocks_syscall() at syscall.c:604)
1680398012 WARNING torsocks[13080]: [syscall] Unsupported syscall number 39. Denying the call (in tsocks_syscall() at syscall.c:604)
1680398012 WARNING torsocks[13080]: [syscall] Unsupported syscall number 39. Denying the call (in tsocks_syscall() at syscall.c:604)
1680398012 WARNING torsocks[13080]: [syscall] Unsupported syscall number 39. Denying the call (in tsocks_syscall() at syscall.c:604)
1680398012 WARNING torsocks[13080]: [syscall] Unsupported syscall number 39. Denying the call (in tsocks_syscall() at syscall.c:604)
1680398012 WARNING torsocks[13080]: [syscall] Unsupported syscall number 39. Denying the call (in tsocks_syscall() at syscall.c:604)
1680398012 WARNING torsocks[13080]: [syscall] Unsupported syscall number 39. Denying the call (in tsocks_syscall() at syscall.c:604)
1680398012 WARNING torsocks[13080]: [syscall] Unsupported syscall number 39. Denying the call (in tsocks_syscall() at syscall.c:604)
1680398012 WARNING torsocks[13080]: [syscall] Unsupported syscall number 39. Denying the call (in tsocks_syscall() at syscall.c:604)
1680398012 WARNING torsocks[13080]: [syscall] Unsupported syscall number 39. Denying the call (in tsocks_syscall() at syscall.c:604)
Assertion 'fclose_nointr(f) != -EBADF' failed at src/basic/fd-util.c:125, function safe_fclose(). Aborting.
Aborted (core dumped)
```

### Location in guide
[Link to the corresponding page and section on <https://raspibolt.org>](https://raspibolt.org/guide/raspberry-pi/privacy.html#ssh-remote-access-through-tor-optional)

### Expected behavior
No core dumped and a successful connection to the raspibolt

### Screenshots

### Environment
 - Hardware platform: OrangePi5
 - Operating system: Armbian Ubuntu 22
 - Version: Ubuntu 22.04.2 LTS

### Additional context
-

twofaktor, 2023-04-04:
Did you install Tor on this Ubuntu 22 computer too?

ywiyogo, 2023-05-04:
> Did you install Tor on this Ubuntu 22 computer too?

yes

dutu, 2023-05-05:
Using torify with SSH can sometimes result in compatibility issues, as you've encountered. An alternative method to use SSH over Tor is to configure the SSH client to use a proxy.

* Create or edit the file `~/.ssh/config`

* Add the following lines to the configuration file, replacing `admin@your_server_address` with the appropriate values for your case
```
Host tor_ssh
  HostName admin@your_server_address
  User admin
  ProxyJump none
  ProxyCommand nc -x 127.0.0.1:9050 %h %p
```

* Save the configuration file and exit the text editor

Now, you can connect to your remote server via SSH over Tor:
```
ssh tor_ssh
```

closed 2023-07-11 by github-actions[bot]