raspibolt/raspibolt #1259: [BUG] wrong instruction to start Homer in "Bonus guide: Homer"

dutu, 2023-04-05:
### Describe the bug

In [Bonus guide: Homer](https://raspibolt.org/guide/bonus/raspberry-pi/homer.html),  Section "[First start](https://raspibolt.org/guide/bonus/raspberry-pi/homer.html#first-start)" the command: `npm run serve` doesn't work anymore.

### Expected behavior

Sections "First Start" and "Autostart on boot" needs to be removed, as there is no "running homer". 

Instead `npm run build` has to be executed, to generate the static html files. This has to be just done before moving the distributable output into the nginx website data folder, in section [Website data folder](https://raspibolt.org/guide/bonus/raspberry-pi/homer.html#website-data-folder).

### Additional context
* The same issue has been reported in #1222
* According to the response on issue ["Homer on Raspibolt"](https://github.com/bastienwirtz/homer/issues/510), there is no `npm run serve` command, the build command generates simple html static files which have to be hosted somewhere (behind a webserver)

dutu, 2023-06-05:
Can any of the maintainers check this one? There is no point keeping a bonus guide that cannot be used.

kristapsk, 2023-06-05:
I don't use Homer myself, can't check.

closed 2023-08-11 by github-actions[bot]