raspibolt/raspibolt #1272: LevelDB fatal error

iTzNikolovich, 2023-05-01:
### Describe the issue
Hi all, after months of synchronization today, following a power outage, bitcoin-cli seems to never start. I hope someone can help me. Below are the logs, thanks in advance.

### Screenshots
[image]

### Environment
 - Hardware platform: [Fujitsu Futro S720]
 - Operating system: [Debian 11]
 - Version: [Debian GNU/Linux 11 (bullseye)]

### Log
[debug.log](https://github.com/raspibolt/raspibolt/files/11365837/debug.log)

twofaktor, 2023-05-01:
Did you follow the RaspiBolt guide? (RaspiBolt is for Raspberry Pi)

iTzNikolovich, 2023-05-01:
> Did you follow the RaspiBolt guide? (RaspiBolt is for Raspberry Pi)

Following raspibolt you can install all you need also on a x86 server... This is linux

iTzNikolovich, 2023-05-01:
Solved with `bitcoind -reindex`. You must have to stop the node and then you can run this command from bitcoin user

closed 2023-05-01 by iTzNikolovich

twofaktor, 2023-05-01:
> > Did you follow the RaspiBolt guide? (RaspiBolt is for Raspberry Pi)
> 
> Following raspibolt you can install all you need also on a x86 server... This is linux

minibolt.info