raspibolt/raspibolt #1273: Unable to connect Sparrow to Private Electrum Server via local connection or Tor

harkousb, 2023-05-01:
### Describe the issue
Stuck trying to connect Sparrow desktop wallet to electrs (https://raspibolt.org/guide/bitcoin/desktop-wallet.html#connect-to-electrs).
bitcoind, electrs, and nginx are running without errors. Indexing completed.

error when trying to connect via local connection:
"Could not connect:

Retries exhausted"

error when trying to connect via Tor:
"Could not connect:

SOCKS: Host unreachable. Check if the proxy server is running."

### Location in guide
(https://raspibolt.org/guide/bitcoin/desktop-wallet.html#connect-to-electrs)

### Environment
 - Raspberry Pi 4 B
 - Operating system: Raspberry Pi OS 64bit
 - Version: Debian GNU/Linux 11 (bullseye)

harkousb, 2023-05-02:
I set log_filters = "DEBUG" in electrs.conf, restarted the service. 

running "sudo journalctl -f -u electrs" while trying to connect:

May 01 16:52:51 raspi electrs[1770]: [2023-05-01T23:52:51.809Z ERROR electrs::server] 6: disconnecting due to failed to send response: "{\"id\":3,\"jsonrpc\":\"2.0\",\"result\":{\"height\":782590,\"hex\":\"0000c020fd4b4b7c34983310f187a14106292c288660df6dbd8e020000000000000000007396eb361ae715d6069048e77e752740930887cef866c21aa003c9be76e2c9a8f01a20643e020617413ea70a\"}}\n"
May 01 16:52:51 raspi electrs[1770]: [2023-05-01T23:52:51.809Z WARN  electrs::server] 6: failed to shutdown TCP connection Transport endpoint is not connected (os error 107)

running "sudo journalctl -f -u electrs | grep :server" while trying to connect:
May 01 16:49:55 raspi electrs[1770]: [2023-05-01T23:49:55.828Z ERROR electrs::server] 5: disconnecting due to failed to send response: "{\"id\":3,\"jsonrpc\":\"2.0\",\"result\":{\"height\":780590,\"hex\":\"00805320795ae2eb6a52f8cf757469b182d31eb850ed1c64153d01000000000000000000f9a40be591ef086d82ffe2740567ff2c2fe69df179cb185de37c558f0236145dabdb0e6481760617400620d6\"}}\n"
May 01 16:49:55 raspi electrs[1770]: [2023-05-01T23:49:55.828Z WARN  electrs::server] 5: failed to shutdown TCP connection Transport endpoint is not connected (os error 107)

zynos, 2023-05-04:
And you checked that the elctrs port is enabled in ufw?
Although it should work with tor anyway..

dutu, 2023-05-04:
Try connecting with Electrum. Electrum worked for me, but I had intermittent issues with Sparrow for some reasons.
That's why I decided to move to Fulcrum.

zynos, 2023-05-04:
Strange, for me its working, i also tested it with the newest sparrow version again..

harkousb, 2023-05-05:
@zynos yes I double checked this step:
$ sudo ufw allow 50002/tcp comment 'allow Electrum SSL'

@dutu I tried connecting with Electrum as well, same problem.

zynos, 2023-05-05:
What is your version of BTC Core? Mine is 23.
You can also have a look here:
https://community.getumbrel.com/t/sparrow-wallet-desktop/6898/24

harkousb, 2023-05-05:
@zynos v24.0.1, I tried switching to 23.1 but had the same issue, so I switched back to 24.0.1 again.

I tried running the [RaspiBolt System Overview](https://raspibolt.org/guide/bonus/raspberry-pi/system-overview.html) and it gets stuck at loading Electrum Server Data
[image]

Running `sudo journalctl -f -u electrs | grep :server` with the log_filters = "DEBUG" in electrs.conf file while trying to connect using Sparrow shows this:
[image]

On the Sparrow side:
[image]

Any ideas what could be causing this? Seems to me like it's an issue with my electrs.

harkousb, 2023-05-05:
It might be because my compaction isn't complete in electrs. I'll try again once that's done.

harkousb, 2023-05-05:
Okay yeah that was the problem. I feel stupid, but hope this helps someone running into the same issue. Once electrs finished compacting it worked.

closed 2023-05-05 by harkousb