raspibolt/raspibolt #1290: [BUG] System Overview Issue

Mynima, 2023-05-27:
### Describe the bug
Latest version update of Bitcoin Core (25.0.0) has broken the system overview guide. Looks like it doesn't recognize the latest version as being up-to-date. As such it is incorrectly flagging to be updated.

### Location in guide
[Link to the corresponding page and section on <https://raspibolt.org>](https://raspibolt.org/guide/bonus/raspberry-pi/system-overview.html)

### Expected behavior
Should recognize latest version as being valid.

### Screenshots
[image]

### Environment
 - Hardware platform: Raspberry Pi 4
 - Operating system: Raspberry Pi OS 64bit
 - Version: Debian GNU/Linux 11 (bullseye)

### Additional context
I'll try take a look into this problem myself if I get chance this week. Otherwise, figured I'd drop it here in case anyone else wants to have a punt.

zynos, 2023-05-27:
Did you installed the latest version of the script (raspibolt-pulse)?
Or are you using the legacy system overview?

Mynima, 2023-05-28:
> Did you installed the latest version of the script (raspibolt-pulse)? Or are you using the legacy system overview?

Hi Zynos, yeah good question. Looks like maybe not. I think last time I updated it was when I was doing the testing for Raspibolt V3. I'll go ahead and upgrade to see if that resolves the issue. 

Funnily enough though I've not had any problems until now. The only thing that changed recently is that I upgraded bitcoin core yesterday while working though the PR.

Mynima, 2023-05-28:
Sorted, it was out of date, thanks zynos.

closed 2023-05-28 by Mynima