raspibolt/raspibolt #1295: Unable to connect to Lightning Terminal by visiting web UI

harkousb, 2023-06-12:
Trying to complete the step to test the connection via web UI in 'Run Lightning Terminal' section of [https://raspibolt.org/guide/bonus/lightning/lightning-terminal.html#upgrade]

"Test that Lightning Terminal is working by visiting the web UI

Paste the following URL in your browser: [https://raspibolt.local:8444/](https://raspibolt.local::8444/) (replace raspibolt.local by your node IP address if required)"

I accepted the SSL certificate, and I see the following:

[image]

litd is running but it looks like I'm getting an error:
[image]

harkousb, 2023-06-24:
Getting connection refused on port 8443 when trying to use loop via CLI.

[image]

Any ideas why this could be?

harkousb, 2023-06-24:
If I manually run litd as user lit, I'm able to run CLI commands.

litd.service is running, status is active, but litd.service isn't automatically running litd?

derekross, 2023-07-21:
I have a brand new install and I'm getting the same error from https:my-host.local:8443

Client sent an HTTP request to an HTTPS server.

 LND Version             0.16.4-beta commit=v0.16.4-beta                               
 LiT Version             0.10.2-alpha commit=v0.10.2-alpha

atamti, 2023-08-02:
I worked through the issue in the NGINX reverse proxy and found this to be the issue: 
Lightning Terminal only deals in https:

https://github.com/lightninglabs/lightning-terminal/blob/master/doc/letsencrypt.md

But the SSL after "listen 8444" in the nginx config in the guide terminates the SSL connection and sends the requests as http.

Location in guide: https://raspibolt.org/guide/bonus/lightning/lightning-terminal.html#firewall--nginx

Removing the SSL resolved the issue for me, as it allows the traffic to simply be passed through encrypted to the lightning terminal app, which will then handle the decryption:

stream {
    upstream litd {
        server 127.0.0.1:8443;
    }

    server {
        listen 8444;
        proxy_pass litd;
    }
}

Recommend devs look into this to see if it's a universal issue and modify the guide accordingly.

harkousb, 2023-08-04:
Worked for me! Thanks!

closed 2023-08-04 by harkousb

dawiepoolman, 2023-08-13:
Yes, worked for me too thx.
sudo nano /etc/nginx/streams-enabled/litd-reverse-proxy.conf

```
upstream litd {
    server 127.0.0.1:8443;
}

server {
    listen 8444;
    proxy_pass litd;
}
```

orangeshyguy21, 2024-06-28:
Great find! Also confirming that this fixes the issue.