raspibolt/raspibolt #1296: I want use Bisq with my node

moisesaer, 2023-06-13:
I want to use Bisq with my node.
I was confused with Bitcoin.conf (whitelist=bloomfilter@192.168.0.0/16****   # for Bisq). on Bonus Section BISQ.
Question: I will use @192.168.0.0/16 or I will use @ X.X.X.X of my local network where is Bisq or @X.X.X.X of my Bitcoin node?
Thanks a lot.
Excuse me,  I am noobie.

twofaktor, 2023-06-14:
`192.168.0.0/16` is a range of IPs of the majority of local network ranges, what local IP address do you have assigned to your node?

twofaktor, 2023-06-14:
> > `192.168.0.0/16` is a range of IPs of the majority of local network ranges, what local IP address do you have assigned to your node?
> 
> Thanks to answer me. My node has ip 10.0.0.40 and range of my local network is 10.0.0.10 until 10.0.0.100. ok?

Then you need change `192.168.0.0/16` to `10.0.0.0/24`

moisesaer, 2023-06-14:
> Then you need change `192.168.0.0/16` to `10.0.0.0/24`

Very good. I undestood. Thanks a lot.

closed 2023-06-14 by twofaktor