raspibolt/raspibolt #1300: Upgrading to Debian Bookworm

barbastrejo12, 2023-06-19:
When is it safe to migrate to debian bookworm? Can we do it now, or better wait a few months?

luksocculta, 2023-06-28:
It isn't safe. Pi OS is little different from main Debian so it isn't recommended to upgrade by Debian way. It's recommended to wait for official upgrade from Raspberry Pi site which probably be done in fee weeks. But it isn't prohibited. It can be successfull or not, you won't know unless you try. It strictly depends from configuration. I wanted to do some kind of Franken-Debian by only pointing repositories to Bookworm but this is also not recommended but many does systems like this and they say it works flawlessly. It's up to you. I give up after some struggles and will wait for official Raspi OS.

barbastrejo12, 2023-06-28:
I see thanks for the info. Better wait then

closed 2023-06-28 by barbastrejo12

purpleKarrot, 2023-11-13:
A version of Raspberry Pi OS based on bookworm was released on 10.10.2023.
Release notes: https://downloads.raspberrypi.com/raspios_lite_arm64/release_notes.txt

hellresistor, 2023-12-17:
I use this imges ..
https://raspi.debian.net/tested-images/