raspibolt/raspibolt #1303: [FEATURE REQ] add the option to backup lnd with gitea instead of github, or on nextcloud.

barbastrejo12, 2023-06-20:
### What & why

Creating an account on github for the backup does not seem a good idea in terms of privacy, so why not on gitea and/or nextcloud that people can easily self-host?

kristapsk, 2023-06-20:
> Creating an account on github for the backup does not seem a good idea in terms of privacy

It's similar with any other centralized service. But recommendation is to use it using Tor only and to create special account for this purpose:

> Go to [GitHub](https://github.com/), sign up for a new user account, or log in with an existing one. If you don’t want for GitHub to know your identity and IP address in relation to your Lightning node, it is recommended to create new account even if you have existing one, and use [Tor Browser](https://www.torproject.org/download/) for this and following steps.

Then part of guide is to configure git for channel backup user to always do anything only via Tor when accessing any git repositories.

> `git config --global core.sshCommand "torsocks ssh"`

But Gitea or anything else using git should be simple, big parts doesn't change, only creating repo and granting access rights.

kristapsk, 2023-06-20:
About self hosting - you should always try to have backups in different physical locations. So it should not be self hosting at the same physical building where your node is located.

barbastrejo12, 2023-06-20:
> About self hosting - you should always try to have backups in different physical locations. So it should not be self hosting at the same physical building where your node is located.

Yes I agree, the idea if to offer more options, but in the end it's the user responsability to mitigate those risks.

closed 2023-08-26 by github-actions[bot]