raspibolt/raspibolt #1305: Upgrading Electrs to 0.10.0-rc.1

luksocculta, 2023-06-25:
### Describe the issue
After command cargo clean there is error:
error: failed to parse manifest at `/home/admin/rust/electrs/Cargo.toml`
Caused by:
  failed to parse the `edition` key

Caused by:
  this version of Cargo is older than the `2021` edition, and only supports `2015` and `2018` editions.

### Location in guide
[Link to the corresponding page and section on <https://raspibolt.org>](https://raspibolt.org/guide/bitcoin/electrum-server.html#for-the-future-electrs-upgrade)

### Screenshots
If applicable, add screenshots to help explain your problem.

### Environment
 - Hardware platform: Raspberry Pi 4
 - Operating system: Raspberry Pi OS 64bit, Debian 11
 - Version: Description:	Debian GNU/Linux 11 (bullseye)

### Additional context
Cargo in Bullseye is 0.47.0-3+b1, in Bookworm is 0.66.0+ds1-1. Do we need wait for Raspberry Pi OS upgrade or can we do some manual intervention? For example: curl https://sh.rustup.rs -sSf | sh.

luksocculta, 2023-06-26:
Ok today i will give a chance and I will change only sources for bookworm. Will see what will happen. I won't do dist-upgrade.

luksocculta, 2023-06-26:
I need to move upgrading for tomorrow as for now I am copying entire blockchain to backup location as don't want to make initial blockchain download again.

luksocculta, 2023-06-27:
i am givin up. i am even unable to make copy of blockchain as i can't stop effectively bitcoind. 'bitcoin-cli stop 'simply doesnt work and bitcoind is starting again automatically.

kristapsk, 2023-06-27:
@luksocculta It's restarted automatically by systemd. Do `sudo systemctl disable bitcoind` and `sudo systemctl stop bitcoind`.

luksocculta, 2023-06-29:
Thanks for detailed guide. Right now I am during making similar configuration of my Raspibolt on different machine and there I will test this. I wanted to simply change sources so thanks about that additional necessary steps.

luksocculta, 2023-06-30:
On a test machine something went wrong with nginx. 

sudo systemctl status nginx
× nginx.service - A high performance web server and a reverse proxy server
     Loaded: loaded (/lib/systemd/system/nginx.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Fri 2023-06-30 23:06:45 CEST; 34s ago
       Docs: man:nginx(8)
    Process: 1378 ExecStartPre=/usr/sbin/nginx -t -q -g daemon on; master_process on; (code=exited, status=1/FAILURE)
        CPU: 22ms

Jun 30 23:06:45 debian-acer systemd[1]: Starting nginx.service - A high performance web server and a reverse proxy server...
Jun 30 23:06:45 debian-acer nginx[1378]: 2023/06/30 23:06:45 [emerg] 1378#1378: unknown directive "stream" in /etc/nginx/nginx.conf:10
Jun 30 23:06:45 debian-acer nginx[1378]: nginx: configuration file /etc/nginx/nginx.conf test failed
Jun 30 23:06:45 debian-acer systemd[1]: nginx.service: Control process exited, code=exited, status=1/FAILURE
Jun 30 23:06:45 debian-acer systemd[1]: nginx.service: Failed with result 'exit-code'.
Jun 30 23:06:45 debian-acer systemd[1]: Failed to start nginx.service - A high performance web server and a reverse proxy server.

But I haven't mimic fully raspibolt configuration as I couldn't perform full blockchain download. But I think that doesn't matter for nginx so right now I dont know what went wrong.

After dist-upgrade I made apt autoremove and then if I good recall was removed something that depends for nginx.

luksocculta, 2023-07-02:
so i made complete upgrade and everything seems to be ok. and there wasn't need for manual installation of libningx-mod-stream as it was installed during upgrade i think. seems like something went wrong on test machine.
only one moment was thrilling. during upgrading of dphys-swapfile terminal hangsout for more that 5 minutes. maybe more. and during all process of upgrading swap, which has 2GB,  was filled to the top . maybe should be doubled for upgrading.

luksocculta, 2023-07-05:
i haven't overwritten any of my configuration files. for all of questions about new config files i answered 'N'. so still everything works normal. sure i still struggle with some things on my raspibolt but these are unrelated to upgrading to bookworm.

luksocculta, 2023-07-06:
ots has broke after upgrading. so i did make 'sudo apt reinstall python3 python3-dev python3-pip python3-setuptools python3-wheel' and then 'sudo pip3 install opentimestamps-client --break-system-packages'.
now works fine.

closed 2023-09-11 by github-actions[bot]