raspibolt/raspibolt #1310: Missing font/symbol

luksocculta, 2023-07-07:
### Describe the issue
Does anybody know why on overview here are missing some symbols?

### Location in guide
[Link to the corresponding page and section on <https://raspibolt.org>](https://raspibolt.org/guide/bonus/raspberry-pi/system-overview.html)

### Screenshots
[image]

### Environment
 - Hardware platform: [Raspberry Pi 4
 - Operating system: Raspberry Pi OS 64bit upgraded to bookworm
 - Version: Debian GNU/Linux 12 (bookworm)

### Additional context
There wasn't those symbols even before upgrading.

luksocculta, 2023-07-17:
No, completely not related to bookworm. It was like that from beginning, I just didn't bother up to now because it's minor issue.

Mynima, 2023-07-17:
> No, completely not related to bookworm. It was like that from beginning, I just didn't bother up to now because it's minor issue.

Hmm, super strange. I didn't have this issue at all on my side with a similar setup (only difference I saw was bookworm, hence the question).

closed 2023-09-22 by github-actions[bot]