raspibolt/raspibolt #1311: [FEATURE REQ] Netdata

Deekee62, 2023-07-09:
### What

Add a guide that allows to install Netdata on Raspibolt.

### Why

Netdata gives far more insight than System overview in the Raspberry Pi Bonus Section

Netdata consists of a daemon that, when executed, is responsible for collecting and displaying information in real-time. It is a lightweight tool, mostly written in C, Python, and JavaScript, which uses minimal resources: about 2% on a single-CPU system.

Netdata is a free and open-source (FOSS) 

### How

A new bonus guide in the Raspberry Pi section, the installation is quite easy and can be installed using a single line installer and with only a few modification afterwards

### Animated GIF

[image]

Mynima, 2023-07-17:
Seems like a 'nice to have' tool, though we need to be careful with adding more projects onto a node. Remember that nodes with funds are hot wallets, users should choose carefully what software they install to minimize this risk where possible.

That said feel free to put together a PR and offer it up. People can choose their own risk profiles.

Deekee62, 2023-07-18:
Unfortunately, I have never made a PR and might need some help!

Deekee62, 2023-07-20:
well, to be honest I've not installed it yet on may RPi Umbrel node, but on an other RPI and the load seems to be minimal...
[image]
[image]

closed 2023-09-25 by github-actions[bot]