raspibolt/raspibolt #1332: [BUG] Tor warning

Deekee62, 2023-08-03:
### Describe the bug
By chance i came across the following error message in the log after a reboot

```
Tor[942]: We compiled with OpenSSL 30000070: OpenSSL 3.0.7 1 Nov 2022 and we are running with OpenSSL 30000090: 3.0.9. These two versions should be binary compatible.
Tor[942]: Tor 0.4.7.13 running on Linux with Libevent 2.1.12-stable, OpenSSL 3.0.9, Zlib 1.2.13, Liblzma 5.4.1, Libzstd 1.5.4 and Glibc 2.36 as libc.
Tor[942]: Tor can't help you if you use it wrong! Learn how to be safe at https://support.torproject.org/faq/staying-anonymous/
Tor[942]: Tor was compiled with zstd 1.5.2, but is running with zstd 1.5.4. For safety, we'll avoid using advanced zstd functionality.
```

When checking the version a warning is displayed too, which seems point to the same issue.

```
`$ tor --version
Aug 03 08:34:33.434 [warn] Tor was compiled with zstd 1.5.2, but is running with zstd 1.5.4. For safety, we'll avoid using advanced zstd functionality.
Tor version 0.4.7.13.
Tor is running on Linux with Libevent 2.1.12-stable, OpenSSL 3.0.9, Zlib 1.2.13, Liblzma 5.4.1, Libzstd 1.5.4 and Glibc 2.36 as libc.
Tor compiled with GCC version 12.2.0`
`
```

### Location in guide
https://raspibolt.org/guide/raspberry-pi/privacy.html#installation

### Expected behavior
- Displaying version
- No error message when checking version

### Screenshots
N/A

### Environment
- Hardware platform: Apple Macmini6,2
- Operating system: Debian 12
- Version: No LSB modules are available. Debian GNU/Linux 12 (bookworm)

### Additional context

Tor was installed quite a while ago using using the raspibolt guide

`$ sudo apt install tor`

It seems the guide has been updated as it's now installed using

`$ sudo apt install apt-transport-https`

### Suggestion

Add an update section for tor

kristapsk, 2023-08-03:
I would try `sudo apt update` and `sudo apt upgrade` and then check if problem still persists.

Deekee62, 2023-08-03:
Removing the old version of tor and installing again didn't help either, so for now I will simply ignore the error.

Obviously, people are aware of this, for more information see:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=963151#14

kristapsk, 2023-08-03:
> Obviously, people are aware of this, for more information see:
> https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=963151#14

This looks like out of the scope of the RaspiBolt guide, we should wait for an upstream fix.

closed 2023-10-09 by github-actions[bot]