raspibolt/raspibolt #1334: How does the backup work?

barbastrejo12, 2023-08-07:
I would like to know more how the backup works? Does it generate a new static-channel-backup every time there's a new tx or routing event?

twofaktor, 2023-08-07:
https://raspibolt.org/guide/lightning/channel-backup.html#why-are-lightning-channel-backups-important

barbastrejo12, 2023-08-08:
> https://raspibolt.org/guide/lightning/channel-backup.html#why-are-lightning-channel-backups-important

is does not answer my questions: Does it generate a new static-channel-backup every time there's a new tx or routing event?

twofaktor, 2023-08-08:
> > https://raspibolt.org/guide/lightning/channel-backup.html#why-are-lightning-channel-backups-important
> 
> is does not answer my questions: Does it generate a new static-channel-backup every time there's a new tx or routing event?

You need to read, it comes in the link that I have given you

closed 2023-10-14 by github-actions[bot]