raspibolt/raspibolt #1341: Add lightning support to mempool block explorer [FEATURE REQ]

raelschiffler, 2023-08-25:
### What & why

Is your feature request related to a problem, or a new addition to the guide? Provide a clear and concise description of what you'd like added to the RaspiBolt guide. If possible, please phrase it as a user story:

> As a user and enthusiast of this project, I want the mempool block explorer tutorial to contemplate lightning network features already present in the project, so that the Raspibolt project becomes even more features-full and complete.

### How

Describe the solution you're thinking of. Provide a clear and concise description of what you want to happen. Did you consider alternative options?

In the backend/mempool-config.sample.json file there's already a sample to do it!
Below is the content:
```
  "LND": {
    "TLS_CERT_PATH": "tls.cert",
    "MACAROON_PATH": "readonly.macaroon",
    "REST_API_URL": "https://localhost:8080",
    "TIMEOUT": 10000
  },
  "CLIGHTNING": {
    "SOCKET": "lightning-rpc"
  },
```
We only have to adapt it to the Raspibolt environment and add to the tutorial!

raelschiffler, 2023-08-25:
Anyway there's a caveat in this request: I think it should contemplate the creation of another macarron file that does not embeds the same full authorizations the admin.macarron has. 
It's food for thought to check what access should it have and more details...
Related: https://github.com/lightningnetwork/lnd/issues/20

raelschiffler, 2023-10-09:
I've found out the lightning explorer only works for paid mempool collaborators.

closed 2023-10-09 by raelschiffler

twofaktor, 2023-11-15:
> I've found out the lightning explorer only works for paid mempool collaborators.

Where did you find this?